Klipper3d/klipper #1750: Can't install

kevimccu, 2019-06-21:
pi@octopi:~ $ git clone https://github.com/KevinOConnor/klipper./klipper/scripts/install-octopi.sh
Cloning into 'install-octopi.sh'...
remote: Not Found
fatal: repository 'https://github.com/KevinOConnor/klipper./klipper/scripts/install-octopi.sh/' not found
This is the error I am getting when trying to install

voodoo-bravo, 2019-06-21:
Hey,
you seem to have merged two commands into one.

Take a look at https://www.klipper3d.org/Installation.html
Every line there is a single command to execute.
```
git clone https://github.com/KevinOConnor/klipper ## press Enter
./klipper/scripts/install-octopi.sh               ## press Enter
```

-Markus

kevimccu, 2019-06-21:
Ahh ok. I thought so. Got a little confused. Thanks

sagonzal, 2019-06-22:
Sorry to bother guys....but I can't install Klipper either, but in my case, the install script fails because it tell me the following:

> ###### Running apt-get update...
> Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
> Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
> Reading package lists... Done
> 
> 
> ###### Installing packages...
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> Note, selecting 'libncurses5-dev' instead of 'libncurses-dev'
> build-essential is already the newest version (12.3).
> libffi-dev is already the newest version (3.2.1-6).
> python-dev is already the newest version (2.7.13-2).
> python-virtualenv is already the newest version (15.1.0+ds-1).
> virtualenv is already the newest version (15.1.0+ds-1).
> virtualenv set to manually installed.
> The following additional packages will be installed:
>   libftdi1 libnewlib-dev libstdc++-arm-none-eabi-newlib libtinfo-dev
> Suggested packages:
>   avrdude-doc task-c-devel gcc-doc ncurses-doc libnewlib-doc
> The following NEW packages will be installed:
>   avr-libc avrdude binutils-arm-none-eabi binutils-avr dfu-util gcc-arm-none-eabi gcc-avr libftdi1 libncurses5-dev libnewlib-arm-none-eabi
>   libnewlib-dev libstdc++-arm-none-eabi-newlib libtinfo-dev libusb-dev stm32flash
> 0 upgraded, 15 newly installed, 0 to remove and 149 not upgraded.
> Need to get 91.7 MB/111 MB of archives.
> After this operation, 848 MB of additional disk space will be used.
> Get:1 http://espejito.fder.edu.uy/raspbian/raspbian stretch/main armhf libftdi1 armhf 0.20-4 [17.0 kB]
> Err:1 http://espejito.fder.edu.uy/raspbian/raspbian stretch/main armhf libftdi1 armhf 0.20-4
>   Hash Sum mismatch
>   Hashes of expected file:
>    - SHA256:023c4bfc54db1218876f57b22c6c1da70d37ba7a253930ee43d33a9a05f36194
>    - SHA1:535d62a96dfb00902a1bef84605da3a599aff8c4 [weak]
>    - MD5Sum:fc8dd106357b0de888620cc8e7fb6397 [weak]
>    - Filesize:16984 [weak]
> 

...and it goes on telling me the there is a Hash Sum mismatch in many many packages.

any ideas ?

problem with mirrors ?

thank you in advance!

kevimccu, 2019-06-22:
Can't get a log file due to the failed install

KevinOConnor, 2019-07-10:
I'm going to close this issue as it looks like the original question was answered (each command must be run individually).

-Kevin

closed 2019-07-10 by KevinOConnor

kevimccu, 2019-07-10:
I got it to install. Don't know why it was still open

sagonzal, 2019-07-10:
I change mirrors and finally I got it right.