Klipper3d/klipper #1752: Printer is not ready

kpayney1, 2019-06-22:
Having issues getting klipper to start. Tried finding the klipper.log file but putty can't find the directory or the file.  This is the octopi log
Connecting to: /tmp/printer
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x6e4fd9b0, open=True>(port='/tmp/printer', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok FIRMWARE_VERSION:v0.7.0-611-gdaadb36 FIRMWARE_NAME:Klipper
Send: M21
Recv: ok
[...]
Send: status
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: // Klipper state: Not ready
Recv: !! Printer is not ready
Recv: ok
[...]
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Closing serial port due to emergency stop M112.
Recv: ok
Connection closed, closing down monitor
Changing monitoring state from "Operational" to "Offline (Error: Closing serial port due to emergency stop M112.)"
Connecting to: /tmp/printer
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x73339df0, open=True>(port='/tmp/printer', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok FIRMWARE_VERSION:v0.7.0-611-gdaadb36 FIRMWARE_NAME:Klipper
Send: N2 M21*18
Recv: ok
[...]
Send: status
Recv: // Shutdown due to M112 command
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: // Klipper state: Not ready
Recv: !! Shutdown due to M112 command
Recv: ok
[...]
Send: firmware_restart
Recv: ok
[...]
Send: status
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: // Klipper state: Not ready
Recv: !! Printer is not ready
Recv: ok
[...]

I have reflashed it 3 times with different baud rates. Using a c10s flashed with the cr10s config file. 
Serial is /dev/serial/by-id/usb-FTDI_USB_UART_AC024EEB-if00-port0. 

Running out of ideas, if anyone could help me that would be great!

kpayney1, 2019-06-22:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3316588/klippy.log)

jakep82, 2019-06-22:
Your indicates you have the wrong serial port configured. 

`Unable to open port: [Errno 2] could not open port /dev/serial/by-id/usb-FTDI_USB_UART_AC024EEB-if00-port0: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-FTDI_USB_UART_AC024EEB-if00-port0'

kpayney1, 2019-06-22:
Thats the port which the /dev/serial/by-id/* lists. Any other ways of identifying the port needed?

kpayney1, 2019-06-22:
doing a lsusb reads me this
https://imgur.com/BNn9auo

kpayney1, 2019-06-23:
Sorted it! When flashing on a cr10-s you need to flash against dev/ttyUSB0 or whatever is listed as the connection on octopi. Using the serial/dev/by-id address doesn't flash the MB for some reason.

closed 2019-06-23 by kpayney1