Klipper3d/klipper #1753: minclock and reqclock

smark-, 2019-06-22:
Digging into the code I found out that the time critical commands are characterized by these two time constraints? What is the exact difference between the two?
I was also wondering if it would be possible to dump the mcu messages sent from the host with their time tag (or clock cycle). Unfortunately with the parsedump.py you get a list of commands but not their time tag.

KevinOConnor, 2019-06-22:
The `minclock` of a command is the minimum time (represented in clock ticks on the micro-controller) that the micro-controller must be at when it receives the command for processing.  The host buffering code (klippy/chelper/serialqueue.c) will buffer commands to ensure they are not received prior to their minclock.  (Commands that are being held in this way show up in the stats as "stalled_bytes".)

The `reqclock` of a command is the "requested clock time" - the worst case time that the command should be received at the micro-controller.  The host buffering code will prioritize commands so that (if their is contention) commands with a lower reqclock will be sent before other commands.  In a sense, minclock is the earliest valid time and reqclock is the maximum valid time of a command.

Command queues (aka cmd_queue) are directly linked with minclock/reqclock.  Commands placed on the same command queue will never be reordered with respect to each other.  However, commands placed on different command queues can have different priorities.  As an example, a PID update can have higher priority than bulk queue_step commands even though the movement commands may have been generated well in advance of the PID update.  Similarly, LCD updates (which are given the lowest possible priority) can be delayed behind all other updates.

It's common to see a command generated with minclock=0 and reqclock=0 - that just means it can be sent at any time (minclock=0) and that it should go out with the highest priority (reqclock=0).

> I was also wondering if it would be possible to dump the mcu messages sent from the host with their time tag (or clock cycle).

Not currently.

> Unfortunately with the parsedump.py you get a list of commands but not their time tag.

The minclock/reqclock are internal to the host buffering code - they aren't transmitted to the micro-controller.  If a command has a `clock` parameter in it, then the reqclock is generally set to that time however.

-Kevin

KevinOConnor, 2019-07-10:
I'm going to close this issue as it looks like the original question was answered.

-Kevin

closed 2019-07-10 by KevinOConnor