Klipper3d/klipper #1754: Feature request: M906 / M907 / M908 support (change motor current on-demand)

elliotf, 2019-06-22:
First, thank you very much for klipper.  :)

Second, I'm not sure if this would be easy or difficult, but I found it useful with my duet wifi (when using sensorless homing) to turn the current to motors down so that they would miss steps more easily.

I'm not seeing a way to do this, currently, as it looks like the current is set in the configuration and that's it.

This seems like it "should" be a reasonably easy feature to implement, as it seems like it should just involve a translation between the format of the M906 / M907 / M908 arguments and existing code that sets motor current.

I'd like to take a crack at a PR, if the above is reasonable.

KevinOConnor, 2019-06-22:
FYI, you should be able to do this now by using a homing_override config section and issuing a SET_TMC_CURRENT before and after the homing request.  See config/example-extras.cfg and docs/G-Codes.md for the details.

-Kevin

elliotf, 2019-06-23:
Awesome! Thank you again!

KevinOConnor, 2019-07-10:
I'm going to close this issue as it looks like the original question was answered.

-Kevin

closed 2019-07-10 by KevinOConnor