Klipper3d/klipper #1756: Endstop phase still 0

50koruny, 2019-06-23:
Hi.
 i have a Anycubic Kossel Linear Plus with Trigorilla and TMC2130 on every axis. 
I have a strange problem with setting up endstop phase.
Two endstops (A and C) working great, but third one (B) is still 0.In range of the 0 to 0.
I have tried a different driver and different endstop pin but still without changes.
When i set phase 0 for this endstop, its going to make error, because a real phase is some different (22, 44 or other.it can vary).
Its a hardware error?
Endstops seems to be working fine without endstop phase settings activated.

Thank you for help. 
 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3318429/klippy.log)

KevinOConnor, 2019-06-23:
What does DUMP_TMC report for each of your steppers?

-Kevin

50koruny, 2019-06-24:
I have reinstalled whole Klipper yesterday and it was fine for a moment.
Stepper B was set to phase 63 after proper endstop calibration process.
Now i have first print finished and printer is in the same error again:

 Endstop stepper_b incorrect phase (got 20 vs 63)

I bet it jump again to another phase if i change that value to 20.

Here is TMC report:

A

========== Write-only registers ==========
IHOLD_IRUN: 0008190c IHOLD=12 IRUN=25 IHOLDDELAY=8
TPOWERDOWN: 00000000
PWMCONF: 00050480 PWM_AMPL=128 PWM_GRAD=4 pwm_freq=1 pwm_autoscale=1
COOLCONF: 00000000
========== Queried registers ==========
GCONF: 00000000
GSTAT: 00000001 reset=1
IOIN: 11000078 DCIN_CFG5=1 DRV_ENN_CFG6=1 DCO=1 VERSION=0x11
TSTEP: 000fffff TSTEP=1048575
XDIRECT: 00000000
MSCNT: 000001a8 MSCNT=424
MSCURACT: 012b007e CUR_A=126 CUR_B=-213
CHOPCONF: 14028384 toff=4 hend=7 TBL=1 vsense=1 MRES=4(16usteps) intpol=1
DRV_STATUS: 810c0083 SG_RESULT=131 CS_ACTUAL=12 stallGuard=1 stst=1
PWM_SCALE: 00000000
LOST_STEPS: 00000000

B

========== Write-only registers ==========
IHOLD_IRUN: 0008190c IHOLD=12 IRUN=25 IHOLDDELAY=8
TPOWERDOWN: 00000000
PWMCONF: 00050480 PWM_AMPL=128 PWM_GRAD=4 pwm_freq=1 pwm_autoscale=1
COOLCONF: 00000000
========== Queried registers ==========
GCONF: 00000000
GSTAT: 00000001 reset=1
IOIN: 11000078 DCIN_CFG5=1 DRV_ENN_CFG6=1 DCO=1 VERSION=0x11
TSTEP: 000fffff TSTEP=1048575
XDIRECT: 00000000
MSCNT: 000002b8 MSCNT=696
MSCURACT: 0197011f CUR_A=-225 CUR_B=-105
CHOPCONF: 14028384 toff=4 hend=7 TBL=1 vsense=1 MRES=4(16usteps) intpol=1
DRV_STATUS: 810c0079 SG_RESULT=121 CS_ACTUAL=12 stallGuard=1 stst=1
PWM_SCALE: 00000000
LOST_STEPS: 00000000

C

 ========== Write-only registers ==========
IHOLD_IRUN: 0008190c IHOLD=12 IRUN=25 IHOLDDELAY=8
TPOWERDOWN: 00000000
PWMCONF: 00050480 PWM_AMPL=128 PWM_GRAD=4 pwm_freq=1 pwm_autoscale=1
COOLCONF: 00000000
========== Queried registers ==========
GCONF: 00000000
GSTAT: 00000001 reset=1
IOIN: 11000078 DCIN_CFG5=1 DRV_ENN_CFG6=1 DCO=1 VERSION=0x11
TSTEP: 000fffff TSTEP=1048575
XDIRECT: 00000000
MSCNT: 000001f8 MSCNT=504
MSCURACT: 0108000a CUR_A=10 CUR_B=-248
CHOPCONF: 14028384 toff=4 hend=7 TBL=1 vsense=1 MRES=4(16usteps) intpol=1
DRV_STATUS: 810c0083 SG_RESULT=131 CS_ACTUAL=12 stallGuard=1 stst=1
PWM_SCALE: 00000000
LOST_STEPS: 00000000

50koruny, 2019-06-25:
OK.I have accidentaly figured that out, when replaced TMC2130 from the extruder to A4988.
I have wrong naming on the servo pins 5 and 11.
Trigorilla 1.1

closed 2019-06-25 by 50koruny