Klipper3d/klipper #1757: Lots of noise from the second z stepper

dralves, 2019-06-23:
Hi All
  
  I need some help, please. After wasting a few hours debugging this I'm still scratching my head.
  
  I've added a second z stepper to my cartesian printer, on an skr v1.3 using tmc 5160, fresh klipper master. When buzzing (and homing) that second z stepper makes a lot of noise but barely turns (lots of noise but no torque). The other steppers work just fine.

  My current thinking is that this must be a software problem, here's why:
- Problem occurs if using other stepper (not a motor problem)
- Problem if using other position (tried as E1 and as E0 on the skr board, so not a board problem)
- Problem occurs when using other cable (not a cable problem)

The log doesn't report an error when buzzing, but here are the TMC dumps for the two motors:
```DUMP_TMC STEPPER=STEPPER_Z1
DWC2:333564.5 - 192.168.86.3 - GET - /rr_gcode?gcode=DUMP_TMC%20STEPPER%3DSTEPPER_Z1
DUMP_TMC STEPPER=STEPPER_Z1
DUMP_TMC STEPPER=STEPPER_Z1
DUMP_TMC stepper_z1
========== Write-only registers ==========
TPWMTHRS:   00000006 TPWMTHRS=6
COOLCONF:   00000000
IHOLD_IRUN: 00060907 IHOLD=7 IRUN=9 IHOLDDELAY=6
TPOWERDOWN: 0000000a TPOWERDOWN=10
========== Queried registers ==========
GCONF:      00000004 en_pwm_mode=1
CHOPCONF:   14410153 toff=3 hstrt=5 hend=2 tbl=2 tpfd=4 MRES=4(16usteps) intpol=1
GSTAT:      00000005 reset=1(reset) uv_cp=1(Undervoltage!)
DRV_STATUS: 80074000 stealth=1 CSACTUAL=7 stst=1
FACTORY_CONF: 0000000e FACTORY_CONF=14
IOIN:       30000052 REFR_DIR=1 DRV_ENN=1 SD_MODE=1 VERSION=0x30
LOST_STEPS: 00000000
MSCNT:      00000008 MSCNT=8
MSCURACT:   00f7000c CUR_A=12 CUR_B=247
OTP_READ:   0000000e OTP_FCLKTRIM=14
PWMCONF:    00000000
PWM_SCALE:  00440083 PWM_SCALE_SUM=131 PWM_SCALE_AUTO=68
PWM_AUTO:   007a00ff PWM_OFS_AUTO=255 PWM_GRAD_AUTO=122
TSTEP:      000fffff TSTEP=1048575
passed: DUMP_TMC
Stats 31267.1: gcodein=191 mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000007 bytes_write=10484 bytes_read=111573 bytes_retransmit=16 bytes_invalid=0 send_seq=1150 receive_seq=1150 retransmit_seq=966 srtt=0.017 rttvar=0.002 rto=0.026 ready_bytes=0 stalled_bytes=0 freq=100003023 heater_bed: target=0 temp=22.5 pwm=0.000  print_time=942.487 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.5 pwm=0.000
Stats 31268.1: gcodein=191 mcu: mcu_awake=0.003 mcu_task_avg=0.000012 mcu_task_stddev=0.000017 bytes_write=10490 bytes_read=111709 bytes_retransmit=16 bytes_invalid=0 send_seq=1151 receive_seq=1151 retransmit_seq=966 srtt=0.016 rttvar=0.005 rto=0.037 ready_bytes=0 stalled_bytes=0 freq=100003028 heater_bed: target=0 temp=22.3 pwm=0.000  print_time=942.487 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.2 pwm=0.000
Stats 31269.1: gcodein=191 mcu: mcu_awake=0.003 mcu_task_avg=0.000012 mcu_task_stddev=0.000017 bytes_write=10496 bytes_read=111825 bytes_retransmit=16 bytes_invalid=0 send_seq=1152 receive_seq=1152 retransmit_seq=966 srtt=0.016 rttvar=0.005 rto=0.034 ready_bytes=0 stalled_bytes=0 freq=100003022 heater_bed: target=0 temp=22.2 pwm=0.000  print_time=942.487 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.3 pwm=0.000
DUMP_TMC STEPPER=STEPPER_Z
DWC2:33567.35 - 192.168.86.3 - GET - /rr_gcode?gcode=DUMP_TMC%20STEPPER%3DSTEPPER_Z
Stats 31270.1: gcodein=191 mcu: mcu_awake=0.003 mcu_task_avg=0.000012 mcu_task_stddev=0.000017 bytes_write=10502 bytes_read=111924 bytes_retransmit=16 bytes_invalid=0 send_seq=1153 receive_seq=1153 retransmit_seq=966 srtt=0.015 rttvar=0.004 rto=0.032 ready_bytes=0 stalled_bytes=0 freq=100003020 heater_bed: target=0 temp=22.3 pwm=0.000  print_time=942.487 buffer_time=0.000 print_stall=0 extruder: target=0 temp=22.3 pwm=0.000
DUMP_TMC STEPPER=STEPPER_Z
DUMP_TMC STEPPER=STEPPER_Z
DUMP_TMC stepper_z
========== Write-only registers ==========
TPWMTHRS:   00000006 TPWMTHRS=6
COOLCONF:   00000000
IHOLD_IRUN: 00060907 IHOLD=7 IRUN=9 IHOLDDELAY=6
TPOWERDOWN: 0000000a TPOWERDOWN=10
========== Queried registers ==========
GCONF:      00000004 en_pwm_mode=1
CHOPCONF:   14410153 toff=3 hstrt=5 hend=2 tbl=2 tpfd=4 MRES=4(16usteps) intpol=1
GSTAT:      00000005 reset=1(reset) uv_cp=1(Undervoltage!)
DRV_STATUS: 80074000 stealth=1 CSACTUAL=7 stst=1
FACTORY_CONF: 0000000b FACTORY_CONF=11
IOIN:       30000052 REFR_DIR=1 DRV_ENN=1 SD_MODE=1 VERSION=0x30
LOST_STEPS: 00000000
MSCNT:      00000008 MSCNT=8
MSCURACT:   00f7000c CUR_A=12 CUR_B=247
OTP_READ:   0000000b OTP_FCLKTRIM=11
PWMCONF:    00000000
PWM_SCALE:  0000003f PWM_SCALE_SUM=63
PWM_AUTO:   000000ff PWM_OFS_AUTO=255
TSTEP:      000fffff TSTEP=1048575
passed: DUMP_TMC
```
Here's my config:

```
# This file contains common pin mappings for the BIGTREETECH SKR V1.3
# board. To use this config, the firmware should be compiled for the
# LPC176x.

########################
# Main configuration
########################

## Basic steppers configuration ##

[stepper_x]
enable_pin: !P2.1 
step_pin: P2.2
dir_pin: !P2.6
step_distance: .0125
endstop_pin: ^!P1.29
position_endstop: 0
position_max: 245
homing_speed: 50

[tmc5160 stepper_x]
cs_pin: P1.17
spi_software_sclk_pin: P0.4
spi_software_mosi_pin: P4.28
spi_software_miso_pin: P0.5
microsteps: 16
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 300
sense_resistor = 0.075
diag1_pin: XDIAG1

[stepper_y]
enable_pin: !P2.8
step_pin: P0.19
dir_pin: P0.20
step_distance: .0125
endstop_pin: ^!P1.27
position_endstop: 0
position_max: 215
homing_speed: 50

[tmc5160 stepper_y]
cs_pin: P1.15
spi_software_sclk_pin: P0.4
spi_software_mosi_pin: P4.28
spi_software_miso_pin: P0.5
microsteps: 16
run_current: 0.950
hold_current: 0.750
stealthchop_threshold: 300
sense_resistor = 0.075
diag1_pin: YDIAG1

[stepper_z]
enable_pin: !P0.21
step_pin: P0.22
dir_pin: !P2.11
step_distance: 0.0025
endstop_pin: probe:z_virtual_endstop
position_min: -2.2
position_max: 250

[tmc5160 stepper_z]
cs_pin: P1.10
spi_software_sclk_pin: P0.4
spi_software_mosi_pin: P4.28
spi_software_miso_pin: P0.5
microsteps: 16
run_current: 0.950
hold_current: 0.750
stealthchop_threshold: 300
sense_resistor = 0.075

[stepper_z1]
enable_pin: !P2.12
step_pin: P2.13
dir_pin: !P0.11
step_distance: 0.0025

[tmc5160 stepper_z1]
cs_pin: P1.8
spi_software_sclk_pin: P0.4
spi_software_mosi_pin: P4.28
spi_software_miso_pin: P0.5
microsteps: 16
run_current: 0.950
hold_current: 0.750
stealthchop_threshold: 300
sense_resistor = 0.075

## Extruder configuration ##

[extruder]
enable_pin: !P0.10
step_pin: P0.1
dir_pin: P0.0
step_distance: 0.00145456437
nozzle_diameter: 0.3
filament_diameter: 1.750
pressure_advance: 0.15
pressure_advance_lookahead_time: 0.010
heater_pin: P2.7
sensor_type: EPCOS 100K B57560G104F
sensor_pin: P0.24
min_temp: 0
max_temp: 300
max_extrude_only_distance: 200

[tmc5160 extruder]
cs_pin: P1.1
spi_software_sclk_pin: P0.4
spi_software_mosi_pin: P4.28
spi_software_miso_pin: P0.5
microsteps: 16
run_current: 1.4
hold_current: 1.1
stealthchop_threshold: 3
sense_resistor = 0.075
diag1_pin: E0DIAG1

## Heatbed configuration ##

[heater_bed]
heater_pin: P2.5
sensor_type: ATC Semitec 104GT-2
sensor_pin: P0.23
min_temp: 0
max_temp: 130

## Additional configuration ##

[fan]
pin: P2.3

[mcu]
serial: /dev/serial/by-id/usb-Klipper_Klipper_firmware_12345-if00

[printer]
kinematics: cartesian
max_velocity: 200
max_accel: 2000
max_z_velocity: 10
max_z_accel: 100
square_corner_velocity: 5.0

###############################################
##  Probing/Mesh
###############################################

[bltouch]
sensor_pin: P1.25
control_pin: P2.0
pin_move_time: 0.250
pin_up_reports_not_triggered: False
pin_up_touch_mode_reports_triggered: False
x_offset: 32.5
y_offset: -1.5
 
[bed_mesh]
min_point: 20, 30
max_point: 200,215
probe_count: 6,6

##############################################
## Cooling/Fans
##############################################

[heater_fan nozzle_fan]
pin: P2.4
heater: extruder
heater_temp: 50.0
fan_speed: 1.0
```

Many thanks in advance

dralves, 2019-06-25:
Attached the log in file form

dralves, 2019-06-25:
I'd like to get to the bottom of this, but not sure where to start looking, any suggestions?
Any pointers to where I should start looking in the code would be very helpful.

KevinOConnor, 2019-06-25:
Your log indicates you aren't running the Klipper software from this repository.  You'd need to reproduce the issue with the pristine software and attach the log from that session in order for us to assist.

That said, your description sounds like a blown stepper driver.

-Kevin

dralves, 2019-06-25:
Hi @KevinOConnor 

  Thanks for taking a look, your help is very much appreciated.
  I'm on master with Stephan3 DWC patch. The patch code didn't seem related, but fair enough, I'll checkout master, retry and re-post the log.
  Regarding the blown driver, that was my first suspicion too, but I rotated the stepper drivers with the other axis and the problem persisted.
  It'd be awesome to have your thoughts on the best way to go about debugging this methodically.

Thanks
David

dralves, 2019-06-26:
Well, replaced the board and the stepper driver and tried a significantly simplified configuration on pristine master as of 06/23 (log attached).

The problem still occurs, but it's not related with the dual stepper axis. I rotated the pins such that the position marker as the E1 extruder on the SKR board is now the x axis, which still has the same problem as before (lots of noise and no torque). Now z and z1 are fine.

It seems the problem only occurs on the E1 position. I'm at a miss as to what the problem might be. It's either a board design bug or a software bug.

Not sure how to proceed. Advice very much appreciated.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3328211/klippy.log)

dralves, 2019-06-26:
Actually I mispoke. The problem does persist, but it is unrelated to the E1 position. In the config above the problem occurs in the E0 position.
Note that I also switched the steppers and the problem persists on a different stepper too.

FWIW I disabled the extruder stepper too in the config above, so can't be related to the number of  steppers.

Getting more and more confused...

dralves, 2019-06-26:
Hum, among the pins for the extra extruder (E1, which I'm using for Z1) are P0.0 and P0.1, which are also used for i2c. Can that be related?

dralves, 2019-06-26:
Ok, this is officially weird. I added an aux board (same hardware, same firmware). Main board has x,y and extruder and aux board has z and z1. Buzzing the stepper the problem still happens to z1, i.e. they all buzz fine except z1.

Log below is master with a single dwc patch. My previous comment showed that was not the source of the problem.

@KevinOConnor any ideas?

[klippy.log](https://github.com/KevinOConnor/klipper/files/3331988/klippy.log)

dralves, 2019-06-26:
As another data point, this still happens even if the stepper is configured as a manual stepper.

dralves, 2019-06-26:
Yet another data point, this still happens if I add yet another z stepper. Both z1 and z2 have the problem now.

jakep82, 2019-06-26:
Any specific reason why you're using different microstep values for Z and Z1?  That seems like an issue.

dralves, 2019-06-26:
@jakep82 thanks for taking a look. that's just because I've been messing about and trying different things. I had fixed it since and the problem persists. There's no danger because the steppers z1 and z2 are not actually screwed in yet..

KevinOConnor, 2019-06-27:
I'm confused on what is being reported.  Earlier you indicated the problem was on the "skr e1" slot (and that z and z1 both worked if other slots on the board were used), but your latest message seems to indicate the problem is with z1 and z2?  (Regardless of the slot?)

-Kevin

dralves, 2019-06-27:
@KevinOConnor I’m confused too :) I thought the problem was the e1 slot, so I added another mcu thinking that would solve it since I’d be using the standard x,y slots, but no luck.
Furthermore the problem also occurs when I make z1 and z2 manual steppers , though in that instance it’s not exactly the same since I can’t configure the tmc5160 sections

Arksine, 2019-06-27:
Since you are running tmc5160s on a SKR 1.3 I assume that you have cut the CLK pin and wired it to GND?  If not it could be causing problem with the driver.

dralves, 2019-06-27:
@Arksine thanks for looking. Yeah did that to all the drivers. Also rotated the drivers and board to rule if I messed that up.

dralves, 2019-06-28:
@KevinOConnor any suggestions on how to debug this further?

KevinOConnor, 2019-06-28:
I'm still not sure what problem is being debugged.  That said, alas, no I don't have much insight.  Double check that you have motor supply provided at all times and that it is clean (the tmc drivers do funky things when motor power isn't there).  Triple check wiring and connectivity.  Triple check pin settings in the config file.  Try to isolate the problem to one specific repeatable event and then use DUMP_TMC to inspect the state of the driver before/after said event.

-Kevin

dralves, 2019-06-29:
Doh!, Turns out I messed up in the initial debugging and there were a couple of cables with the wires crossed which I didn't track properly.
At least I understand the klipper code base a lot better :)
Thanks for looking.

closed 2019-06-29 by dralves