Klipper3d/klipper #1761: Multi MCUs, same axis?

miked63017, 2019-06-25:
Is it possible to control 2 MCUs on the same axis, for example the z axis on a prusa? 

Might be trying a crazy idea, and I know I could put them on a single MCU, but need to figure out if I can run two separate MCUs for the same axis.

timmit99, 2019-06-25:
Kevin, correct me if I am wrong but:
With Klipper the endstops must be on the same MCU as the drive motors so I don't think it's possible to have them split across unless you can find a way to define an endstop on both MCU's
Klipper wants to have the endstops on the same so it can immediately stop the motors on endstop triggering.

miked63017, 2019-06-25:
@timmit99 , does that mean if I connected an endstop to each MCU I would be able to define 1 axis to use multiple MCUs at the same time?

timmit99, 2019-06-25:
> @timmit99 , does that mean if I connected an endstop to each MCU I would be able to define 1 axis to use multiple MCUs at the same time?

THAT i am not sure about. I have a VORON 2 that is running Klipper and we use 4 Z-steppers all one one ramps shield and a separate ramps for the A-B motors and extruder.

KevinOConnor, 2019-06-25:
As mentioned, the endstop for a stepper must be wired to the micro-controller that controls that stepper.  (And similarly, any Z-probe would need to be wired to all the micro-controllers that control Z steppers.)  If you can satisfy that requirement (either by deploying two Z endstop switches, or by wiring the same Z endstop switch to both micro-controllers) then it is possible to use multiple micro-controllers on a single axis.

That said, be aware that issue #381 describes an issue with homing a long Z axis across multiple microcontrollers.  I hope to have that resolved in the next couple of months.

-Kevin

KevinOConnor, 2019-07-10:
I'm going to close this issue as it looks like the original question was answered.  (Also, FYI, PR #1766 should address the homing issue mentioned earlier.)

-Kevin

closed 2019-07-10 by KevinOConnor