Klipper3d/klipper #1762: Make configs work with symlink

kwint, 2019-06-25:
I have recently learned about dotfiles and symlinks. I have the printer config in my dotfiles repository and symlinked it to the `~/printer.cfg` location. This works fine until I run the `SAVE_CONFIG` command (for example), since that command changes the config it, it think, renames the file to `printer_date.cfg` and makes a new `printer.cfg` file. This breaks the symlink. 

Is it possible to change this behavior?

Hywelmartin, 2019-06-27:
you could have a klipper printer.cfg that looks like this

`[include my_printer.cfg] `
then all save data will be stored in printer.cfg and your symlinked will be untouched

KevinOConnor, 2019-06-28:
As indicated, an [include] may help.  The software uses a rename action to backup the config as that is the safest mechanism to ensure a proper backup.  I don't think we wish to change that approach.

-Kevin

KevinOConnor, 2019-07-10:
I'm going to close this issue as it looks like the original question was answered.

-Kevin

closed 2019-07-10 by KevinOConnor