Klipper3d/klipper #1764: Skew Correction and Homing overide

BlackStump, 2019-06-28:
@Arksine 
I have enabled skew correction but I can not run home all axis ie G28 it errors

Must home axis first: 0.006 0.068 10.227 [0.000]
If I run G28 it errors
If I run G28 Z0 it errors
If I run G28 X0 Y0 that works then I can run G28

I presume it doesn't like the any Z movement first ie a problem with homing override blind Z lift?

Is this expected behaviour with skew enabled?

Arksine, 2019-06-28:
Yes, skew_correction.md mentions this in the Caveats section.  Attempting to move one axis will likely force a correction on the other non-homed axes.  You can get around this by setting the x and y positions in homing_override like you have z.

Also, much like how bed_mesh is capable of moving the tool below Z=0, skew_correction is capable of moving the X and Y axes to positions less than position_min and greater than position_max, depending on how its skewed.   If you find this is happening you can set the position_min and position_endstop of your x and y axes to a negative number, and avoid plating parts too close to the maximum edges.

BlackStump, 2019-06-28:
@Arksine 
Thanks for the clarification and advice, I changed the homing override to include xyz positions and G28 works as expected.

closed 2019-06-28 by BlackStump

Arksine, 2019-06-28:
No problem.  FWIW I am testing a change suggested by Kevin that will help resolve this issue.  Going forward configuration wont be done in printer.cfg, it will be done via gcode.  Recommended usage will be to set the skew in your start gcode, after printing and any actions that move the tool near the edges such as a purge...then clear the skew in the end gcode.  This will prevent skew correction from interfering with homing or manual movement outside of a print.

BlackStump, 2019-06-29:
that is great I look forward to that change, thank you