Klipper3d/klipper #1765: Klipper sometimes seems to run gcode out of order

Lissanro, 2019-06-28:
I have printed some parts successfully with Klipper. Works great (except this particular issue). I have delta printer so I decided to try print at higher speeds and see what happens. However, when I have tried to increase printing speed, jerk (20mm/s for printing and 30mm/s for travel) and accelerations (up to 5m/s², inner walls 3m/s², outer walls 1.5m/s²) Klipper seems to print at expected speed but in OctoPrint it looks like it prints at impossible speed and job finishes very fast before few layers are actually printed. However inability to monitor real progress of the job when this bug is triggered is not the main issue.

It seems that the klipper sends "ok" for all gcode lines before actually finishing the moves (like it tries to buffer all gcode at once). The biggest issue is that temperature related gcode seems to be executed out of order as soon as it is gets in the buffer.

For example, bed temperature for first layer set to 110, but it starts to drop to 90 almost at the beginning of printing first layer. Few layers later the end of gcode gets buffered where are commands to disable heaters so Klippers tries to print remaining layers with "cold" extruder.

MCU load seems to be less than 10% at all times. I use Maple Mini (STM32F103) and RAMPS with DVR8825 drivers set to 32 microstepping. I use stepper motors with 400 steps/revolution.

I have attached klippy.log, gcode that causes the issue, serial log from octoprint and my config file. I also have attached gcode which does NOT cause the issue - it has the same models with the same layer height but different speed/jerk/acceleration settings (when I print no_issue gcode I can monitor progress in octoprint or serial.log at normal rate and everything prints as expected).

The issue is apparent even at the beginning of the first layer (because Klippers sets bed target temperature to 90 degrees long before actually finishing first layer at 110 degrees) so I assume that some gcode at the beginning triggers the bug. Both problematic and no_issue gcode files run similarly while waiting for hotend and bed to reach target temperatures, the issue is triggered after that.

Expected behavior is that gcode should never be executed out of order (or is there some other explanation?) and if temperature drops too low for whatever reason it should abort. I still can use LCD menu when this bug is triggered but cannot set temperature manually (while this issue is triggered) so I have no choice but to press reset button to stop printing otherwise Klipper will continue printing with cold extruder. Perhaps if I remove gcode to disable heaters at the end it could print successfully (if it indeed buffered the whole thing), please let me know if I should try this or anything else.

For me this bug is 100% reproducible - if I try to print attached gcode then print will fail as described above. If I print no_issue gcode it is always successful.

[image]
[klippy.log](https://github.com/KevinOConnor/klipper/files/3339498/klippy.log)
[Backlash_Adapter_5mm.gcode.zip](https://github.com/KevinOConnor/klipper/files/3339520/Backlash_Adapter_5mm.gcode.zip)
[Backlash_Adapter_5mm.no_issue.gcode.zip](https://github.com/KevinOConnor/klipper/files/3339521/Backlash_Adapter_5mm.no_issue.gcode.zip)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/3339544/printer.cfg.txt)

KevinOConnor, 2019-06-28:
There was a recent bug fix (commit 61fc57e6) that appears to be related to your report.  Could you upgrade to the latest code and see if the issue persists?  (See the FAQ for upgrade details.)

-Kevin

Lissanro, 2019-06-28:
I thought I did upgrade to latest code but now I double-checked and it looks like I updated the firmware but forgot to run install-ubuntu-18.04.sh script to update Klipper on my system after recompiling. This was my first experience updating Klipper so I tried to follow all steps carefully but somehow managed to miss this final step.

I confirm that this bug is fixed so I'm closing this issue. I tried again and parts printed very good and quickly. Thank you very much for quick reply and helping me quickly to solve this problem.

closed 2019-06-28 by Lissanro