Klipper3d/klipper #1769: bltouch section incomplete implementation

aschor, 2019-06-29:
the doc says that, for section [bltouch], you can define those parameters : 

[bltouch]
#samples:
#sample_retract_dist:
#samples_result:
#   See the "probe" section for information on these parameters.

=> as a matter of fact, that's wrong. When I try thoses (because I want to control the height the bltouch will retract between the homing probes, to not hit the bed), this is the error I get : 

 Option 'sample_retract_dist' is not valid in section 'bltouch'
or 
 Option 'samples' is not valid in section 'bltouch'
[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/3341703/klippy.log.gz)

BlackStump, 2019-06-29:
have you read this?
https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md
have you updated to the lastest klipper?
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-upgrade-to-the-latest-software

aschor, 2019-06-29:
Hi. 
My klipper setup is very new. I'd say 2 weeks.

I'll conter check if I did setup it 1 day before the options moved to the probe section, that would be no luck.

I had checked the date of the last change of config-extra, but I must have seen the wrong date (something near January).

aschor, 2019-07-01:
Aaaannnnd ..... You were right ! I did checkout in June 4th..... The parameters moved from section around June 6th.... That's unfortunate ! Thank you !
Closing...

closed 2019-07-01 by aschor