Klipper3d/klipper #1770: started to get mcu shutdown for some reason

zmrde, 2019-06-29:
Hello everyone. I've been using klipper from 19 may and suddenly I start to get mcu shutdowns randomly. I haven't changed my printer.cfg.
I have been using klipper version from 17th may and yesterday I pulled the newest version. I upgraded some packages on my rock64 armbian system and started to print. I came back and the print was stopped in the middle (This was the first time I experienced something like this). So I did a hard reset to go back to the 17th may klipper version to see if there is some difference...and it keeps just shutting down randomly. (I reflashed the firmware between versions)

I am using MKS Gen L with rock64 (raspberry pi like pc) and Ender-3 and octoprint

I uploaded the latest logs.

Thank you for help

zmrde, 2019-06-29:
It seems like it was octoprint issue. Switched to dwc2 and I'll test it more. I'll keep you updated.

KevinOConnor, 2019-07-10:
I'm going to close this issue as it looks like the original poster found a solution to their problem.

-Kevin

closed 2019-07-10 by KevinOConnor