Klipper3d/klipper #1773: TMC2130 issues !! Failed to home X: Timeout during endstop homing

scaryman1987, 2019-06-30:
OK so a4988's work fine but none of my tmc2130 work.

wiring is correct and works fine in marlin1.1.9 with exactly the same config

X and y axis doesnt move at all

!! Failed to home X: Timeout during endstop homing

if I trigger x to emulate movement of the X axis the Y does the same

NB: in the log file below I made some mistakes on my endstop configs but those are fixed now
and my end stops trigger normally had to add ^! before the pin

Please help

[klippy.log](https://github.com/KevinOConnor/klipper/files/3343381/klippy.log)

FHeilmann, 2019-07-01:
your config only defines the virtual enables, are you sure that your board doesn't, in addition, have a physical enable pin to the TMC2130? Those have to be configured as well (the virtual enable just allows to switch off individual drivers if they share an enable line), just put the physical enable pin in front of the virtual enable and separate them with a comma.

scaryman1987, 2019-07-01:
Hi FHeilmann - Thank you for assisting - just at work atm. Will check when i get home. but as you said I only had the virtual pin since I assumed you use the virtual pin for the tmc in the same way you use the virtual pin for the z endstop. I will try what you suggested. although i though that tmc's aren't connected to the enable pin where they plug in on the board however I never double checked this and realize now that I could be wrong. What you are saying does make a lot of sense if my original assumption is wrong.

Will update this after testing .
Really hope this is the issue.

scaryman1987, 2019-07-01:
FHeilmann you sir are a star! issue resolve - had to omit the virtual pin all together but its working now.

Still messing with my t-off times to get the idle whine out of these crappy anet a8 steppers but at least i can run a print now. posted an example of the X-axis below this is for a ramps/am8

[stepper_x]
step_pin: ar54
dir_pin: ar55
#enable_pin: !ar38
enable_pin: !ar38
step_distance: 0.01
endstop_pin: ^!ar3
#endstop_pin: ar2
position_endstop: 0
position_max: 220
homing_speed: 40

[tmc2130 stepper_x]
cs_pin: ar53
spi_software_sclk_pin: ar52
spi_software_mosi_pin: ar51
spi_software_miso_pin: ar50
microsteps: 16
interpolate: True
run_current: 0.650
hold_current: 0.3
sense_resistor: 0.110
stealthchop_threshold: 0
driver_TOFF: 2

closed 2019-07-01 by scaryman1987