Klipper3d/klipper #1775: SKR V1.3 serial (AUX-1) connection failure

b-vs, 2019-07-01:
**Issue:**
Can't connect to SKR V1.3 over serial (AUX-1) interface.

**Problem description:**
UART baudrate at SKR V1.3 side is 1.25 times lower than it supposed to be.
This is caused by the missing **FDR** initialization of the **LPC_UART0** at **serial_init()**.

**Solution:**
**serial_init()** at **src/lpc176x/serial.c** should look like:
...
serial_init(void)
{
    // Setup baud
    LPC_UART0->LCR = (1<<7); // set DLAB bit
    enable_pclock(PCLK_UART0);
    uint32_t pclk = SystemCoreClock;
    uint32_t div = pclk / (CONFIG_SERIAL_BAUD * 16);
    LPC_UART0->DLL = div & 0xff;
    LPC_UART0->DLM = (div >> 8) & 0xff;
    **LPC_UART0->FDR = 0x10; // DivAddVal = 0, MulVal = 1**
    LPC_UART0->LCR = 3; // 8N1 ; clear DLAB bit
...

KevinOConnor, 2019-07-11:
Thanks.  Should be fixed now (commit b77375e9).

-Kevin

b-vs, 2019-07-12:
Just tested. Everything seems to be fine now. Thanks.

closed 2019-07-18 by KevinOConnor