Klipper3d/klipper #1777: Adjust bed_mesh points?

dertester, 2019-07-01:
Hey, 

im testing Klipper on my Anycubic Chiron.

I tried mesh bed leveling, is there a way to change all points afterwards besides manually changing every value?

Havent found a command in the docs.

schnello, 2019-07-02:
What Kind of Probe you use?

dertester, 2019-07-02:
I used a basic switch. Its the autoleveling switch that came with the chiron.
It works for PROBE_CALIBRATE or Mesh leveling..

Should i try the value i got from [PROBE_CALIBRATE] ?
I think its not clear what that value does, is it the endpoint?

I dont want my switch pressed to hard or crash again :)

schnello, 2019-07-02:
You should use;
PROBE_ACCURACY SPEED=1 X=10 Y=10 Z=40

>I dont want my switch pressed to hard or crash again :)
Press manualy the switch. If the printer doenst stop (speed=1 ist very slow)... turn of the printer.

dertester, 2019-07-02:
I will try it and post an update. Thank you so far!

Hywelmartin, 2019-07-05:
This setting in the
```
 [probe] 
....
#sample_retract_dist: 2.0
....
```

Is the distance an absolute or relative, the height for movement between points in for example Delta calibration is Absolute Z-height... Confusing at least..

Hywelmartin, 2019-07-06:
If you run PROBE_ACCURACY it will be from Z=10 at current X and Y

Try PROBE_ACCURACY Z=30

https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#probe

closed 2019-07-08 by klipper-gitissuebot