Klipper3d/klipper #1779: correct z height for z min endstop.

mrcpu99, 2019-07-02:
WOrking my way though klipper, and the question is this.

Should I basically get my z endstop set as close to exactly even with the top of my bed, so that when the nozzle is touching the bed, the z endstop fires?

Then I can set up klipper to use the probe leveling, but on the off chance that a bit of the bed is low, how do I allow klipper to take Z below 0.0?

Or should I set the Z min endstop 2mm lower than the bed, so that the z-offsets from the probing are always positive?  Well, maybe that's a negative offset, I read that probe calibration document a dozen times, and I'm still sorting through it.  The X and Y offsets make sense, but the combinatino of all the Z stuff is still a bit of a mystery.

Thanks for your time.

Hywelmartin, 2019-07-02:
in [printer] add

`minimum_z_position:-5.0`

@KevinOConnor . Is this a delta only thing?? the variable is only found in example-delta.cfg not in example.cfg, example-winch.cfg, example-corexy.cfg or example-polar.cfg

otherwise use a virtual endstop([probe as endstop](https://github.com/KevinOConnor/klipper/blob/e19a41d0dd7976617447b4b80b52dfc991643914/config/example-extras.cfg#L11))

KevinOConnor, 2019-07-03:
@Hywelmartin - the equivalent on printers with a cartesian Z is stepper_z position_min.

-Kevin

closed 2019-07-09 by klipper-gitissuebot