Klipper3d/klipper #1780: Z height possible cumulative error

BlackStump, 2019-07-03:
I am not sure how to diagnose this, but here it is I printed a tall print 398mm and a kink occurred around 190mm then the print continued reasonably straight until just before it finished at around 395mm and another kink occurred (pics to follow)

At first I thought it was a hardware problem but scratching my head at what, I had printed a 280mm print before and no probs, I did some basic checks for anything that might be loose but nothing found.

It is a delta printer well built ( but everyone says that ) and not my first printer and not my first Delta it has a build height of 473mm

The print I did I was using S-curve as I have been using S-curve for months.
[image]

The kink is very obvious so how to diagnose whats wrong, no good trying to reprint that as it was a 7 hour print, so made a simple tubular model that was 410mm high

retested with S-curve then did a clean install of klipper straight from the git and here is the 2 prints

[image]

The one on the right is the S-curve and the kink matches the first print at around 190mm so cancel print and started the non s-curve one on the left
when it got past the 190mm mark I thought oh it has something to do with S-curve but then a kink occurred again but this time higher on the print  around the 300mm mark. 

Is it hardware, I am thinking not as why would the location change is it just S-curve, I am thinking not as it happens on master as well, is it a cumulative error, I have no clue.
 
any thoughts ?
@KevinOConnor 

[klippy.log](https://github.com/KevinOConnor/klipper/files/3352973/klippy.log)

BlackStump, 2019-07-03:
Update
I did 2 more prints with a different model as the 15mm tubes have a small foot print on the bed and start to wobble as it gets higher.
so 40mm tube with a 60mm base still single perimeter print above the base ie hollow

2 prints one without S-curve one with S-curve no retraction on both
[image]

 one with the color change is no S-curve the all black is with S-curve 
[image]
 
I am not sure what else to do to quantify this it seems clear there is a problem with tall prints 410mm tall are the height of the prints

KevinOConnor, 2019-07-03:
Seems mechanical in nature to me.  Perhaps the printed part itself is bending slightly at the higher layers.  When s-curves are enabled, it creates higher absolute acceleration, so it can place more stress on the motors and create more vibrations.

FWIW, during a print, the Klipper delta kinematics are defined via just these four lines of code: https://github.com/KevinOConnor/klipper/blob/master/klippy/chelper/kin_delta.c#L20 .  It looks like you have bed_mesh enabled, so that adds a little more complexity.  But, it seems unlikely that it is related.

-Kevin

BlackStump, 2019-07-03:
Thanks for your reply, I am using a 0.6 nozzle printing quite slowly on the tests, If it is mechanical my head keeps saying the location would be static and reproducible, I need to have a closer to look at that side to either find a problem or rule it out.
[image]

bruce356, 2019-07-04:
Hi Just a thought, is it possible that your steppers are missing steps (micro steps) for some reason causing the offset.
This can happen with TMC drivers in stealthchop mode, but of course this may not apply in your case.
regards - bruce

BlackStump, 2019-07-04:
That actually might be the reason as I am using stealthchop, I have changed to spread and retesting atm
thanks for your input much appreciated. 
TMC2100 first version of stealthchop on a problematic board which is more likely the problem then anything else.
I really should hook the board out and replace it, it has been nothing but a saga of problems.

BlackStump, 2019-07-07:
@KevinOConnor 
I really don't have a answer to why but I can say for certain it is not mechanical, It has been bothering me to find the answer.
If it was mechanical and on the surface it does look like mechanical but the only place I can think of that would cause a print to deviate off the printing line and then return back to the correct print line is kinks in the 2040 extrusion.
Problem one the kinks are not always at the same height and there is no visible kinks in the extrusion.
If it was steppers or drivers if it missed or skipped steps it would not return to the print line, this is open loop.

So in frustration
I have not touched anything mechanical at all but what I did was removed the Replicape with Klipper and replaced it with the Duet2 that I had ready to replace the Replicape.
Interesting comparison the Duet2 with RepRapFirmware printed the 398mm tall 2040 with no problems to speak off so same mechanics just different MCU and different Firmware.

Print time with same gcode was approx 2 hours faster with the Duet,
 5 hours vs 7 hours with Replicape/Klipper  but that is not really important for me what is important there is no kinks in the print.

[image]

going from left to right 
scurve-klipper ,  master-klipper  ,  duet-rrf

KevinOConnor, 2019-07-07:
> If it was mechanical and on the surface it does look like mechanical but the only place I can think of that would cause a print to deviate off the printing line and then return back to the correct print line is kinks in the 2040 extrusion.

For a printed part that tall and narrow, even a slight change in the printed part's position (or in the position of the bed) could cause the problem you see.  If the printed part sways a little, the toolhead will continue to deposit filament, but it would no longer line up with the lower layers.  Thus, even slight vibrations could cause corruption of the print.  Slight changes in extrusion could also have that impact  - if the nozzle drags on the print even a little, then the nozzle could drag the printed part out of position.

> Print time with same gcode was approx 2 hours faster with the Duet,

FYI, Klipper calculates stepper movements according to its velocity, acceleration, and look-ahead rules.  If there is a difference in print time, it's due to these configuration settings.  (Assuming there isn't an octoprint style "print stall", which your earlier log does not indicate.)  You should obtain identical print times with Klipper on a ramps, Klipper on a replicape, Klipper on a duet, etc. .  The timing of prints in Klipper is based solely on its look-ahead calculations, and it is not dependent on the particular hardware (or even the type of kinematics).

-Kevin

BlackStump, 2019-07-07:
@KevinOConnor 
I dont use Octoprint for me Octoprint =bloatware yes fully featured but not efficient.
I have not explore why there was a big difference in print time and as you said most likely configuration differences even tho I tried the match acceleration figures but as I said that part was not important to me and with configuration changes  I am sure I can match print times.

I was using the slicers Z hop because it is a tall print on a small foot print, and the print time was not klipper vs klipper it was klipper vs reprapfirmware that comes with the duet2 it was a observation not a criticism of klipper just to be clear.

FYI the bed is fixed to the base so can not move the carriages are 4 wheels and adjusted to be no movement but not binding, the diagonal rods are magnetic n52, if you try to wobble the effector when fixed at a location it is solid, as I said this is not my first Delta so I do know what has been problem area's in previous owned delta's any way I think this conversation is mute.

while I will not dispute what you have said about nozzle dragging print out of alignment, I am not convinced in this case that is the situation, z hop for one with allow print to return to position and when I was watching it print the kinks the print part movement would have been seen.

unfortunately It seems I will be using RepRapFirmware on the Delta with the Duet2 and I will continue to use Klipper only on the Ender3 

I like Klipper very much but I also need confidence that if I want to print tall I will get a successful print

closed 2019-07-07 by BlackStump

KevinOConnor, 2019-07-08:
I understand.  My previous posts were only intended to be informational.

My best guess is that there is not a direct software issue here.  However, it is quite possible that the software is doing something that is indirectly leading to a secondary issue.  It's difficult to guess what that may be though.

If you'd like to investigate Klipper further, my suggestion would be to try and simplify the comparison.  For example, go back to pristine code (no external dwc2 code; no s-curves), disable bed mesh, install Klipper on the duet2, and then reprint.  That is a lot of work, so I understand if that is not something you wish to pursue.

Cheers,
-Kevin

BlackStump, 2019-07-08:
sometime in the future I may do exactly that but for now I need to get these tall prints out.

BlackStump, 2019-07-11:
Rest assured it is not Klipper as RRF is producing similar results, something to do with using 0.6 nozzle.
I need to spend some time trying to figure out what is causing this ...it is not firmware.
seems to be inconsistent in the results.

BugsonPL, 2021-06-14:
@BlackStump did you find out what was the cause?
I'm having very similar issue.

BlackStump, 2021-06-14:
no sorry I did not, I have not printed anything that high since, I suspect it was a layer height thing when using a large nozzle.