Klipper3d/klipper #1781: Anycubic Kossel Linear Plus with detachable Z-Probe Config

Sentinel-sys, 2019-07-03:
Hello, everybody,
this is my first entry here. So please be patient with me...
I have an Anycubic Kossel Linear Plus 3D printer with a removable Z-Probe. Since now two days I try to get this to work in Klipper, unfortunately without success.
But I have to admit that the functions in Klipper are a bit too much for me and the learning curve is very steep.
I also have the feeling that due to the rapid development of Klipper some texts or help files are outdated and don't work anymore.
Maybe someone is using this printer with working autoleveling and Klipper and could help me?
It might also be good to include this model in the config examples.
I am also available as a beta tester. Unfortunately I'm more the mechanic than the programmer... :-)

Many thanks

JAD

Here is my attempt so far:

[stepper_a]
step_pin: ar54
dir_pin: !ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: ^ar2
homing_speed: 60

position_endstop: 292
arm_length: 271.50

[stepper_b]
step_pin: ar60
dir_pin: !ar61
enable_pin: !ar56
step_distance: .0125
endstop_pin: ^ar15

[stepper_c]
step_pin: ar46
dir_pin: !ar48
enable_pin: !ar62
step_distance: .0125
endstop_pin: ^ar19

[extruder]
step_pin: ar26
dir_pin: !ar28
enable_pin: !ar24
step_distance: 0.010989
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar10
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog13
control: pid
pid_Kp: 25.349
pid_Ki: 1.216
pid_Kd: 132.130
min_extrude_temp: 150
min_temp: 0
max_temp: 275

[heater_bed]
heater_pin: ar8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog14
control: watermark
min_temp: 0
max_temp: 130

[fan]
pin: ar9
kick_start_time: 0.200

[heater_fan extruder_cooler_fan]
pin: ar44

[mcu]
serial: /dev/ttyUSB0
pin_map: arduino

[printer]
kinematics: delta
max_velocity: 500
max_accel: 3000
max_z_velocity: 200
delta_radius: 115

minimum_z_position: -5

[idle_timeout]
timeout: 360

[delta_calibrate]
radius: 115
manual_probe: false

[display]
lcd_type: hd44780
rs_pin: ar16
e_pin: ar17
d4_pin: ar23
d5_pin: ar25
d6_pin: ar27
d7_pin: ar29
encoder_pins: ^ar31, ^ar33
click_pin: ^!ar35
kill_pin: ^!ar41

[probe]
pin: ar18
x_offset: 0.0
y_offset: 0.0
z_offset: 15.5
speed: 5.0
samples: 3
sample_retract_dist: 2.0
samples_result: average
samples_tolerance: 0.100
samples_tolerance_retries: 0

#activated_gcode:
#   A list of G-Code commands to execute prior to each probe attempt.
#   See docs/Command_Templates.md for G-Code format. This may be
#   useful if the probe needs to be activated in some way. Do not
#   issue any commands here that move the toolhead (eg, G1). The
#   default is to not run any special G-Code commands on activation.
#deactivate_gcode:
#   A list of G-Code commands to execute after each probe attempt
#   completes. See docs/Command_Templates.md for G-Code format. Do not
#   issue any commands here that move the toolhead. The default is to
#   not run any special G-Code commands on deactivation.

Sentinel-sys, 2019-07-03:
Okay,
now I´m a little bit further. I got the Z-Probe to work and tried some Probe Scripts (PROBE; PROBE_CALIBRATE; DELTA_CALIBRATE; etc.). After all the trying I started from scratch made only a PROBE_CALIBRATION and then printed an calibration cube.
Z axes is okay but X and Y axes are to small. To be presice 15mm instead of 20mm.

See Klipper Log attached.

Regards

[klippy.log](https://github.com/KevinOConnor/klipper/files/3356168/klippy.log)

cmock, 2019-07-07:
If your X/Y sizes are off, then your arm_length is wrong. The best thing from my experience is to do the enhanced delta calibration as described in the docs; it is some effort to measure all these dimensions, but it is worth it, you'll probably never have to do it again unless you change the mechanical structure of your printer.

The basic delta calibration will not help you here as it does not calculate the arm_length.

james-golden, 2019-07-08:
Cmock is correct about the enhanced delta calibrate. I have the same printer and after doing the full calibrate with the stl file, sizes came out almost perfectly

Sentinel-sys, 2019-07-09:
Okay thank you for your help. I will try it this evening after work. But one more question. Do you have the same printer? Anycubic Kossel linear Plus Second Edition with detachable Z-Probe? Then I would kindly ask for your config file, cause I still have some trouble understanding the config options. It would help to see an working config file.
Thank you very much...

cmock, 2019-07-09:
I have a Kossel mini built from a "Sintron" kit, with a self-built detachable probe (a microswitch that clamps to the nozzle). with TMC2130s and a switchable power supply, so I don't know whether my config will do you any good -- and I don't have access to it until friday...

Sentinel-sys, 2019-07-09:
Oh, stepper driver changing is the next thing I will do at my printer. So this config is also interessting for me. :-)

Paulbsh, 2019-07-10:
In all honesty:
bin the probe and go for manual delta calibration. 
It's WAY more reliable.
I did the same with my Anycubic Kossel.

cmock, 2019-07-14:
Here we go, the klipper config. The printer is a Kossel Mini Sintron kit, RAMPS1.4 with TMC2130 for A/B/C, a detachable probe (microswitch that gets clamped to the nozzie) and an ATX-style power supply; it also has a 24V stepup that powers the steppers only, take this into account when looking at the TMC2130 parameters.
The "homing_override" section is to compensate the not-so-perfect probe z_offset.

Hope this helps,

cm.
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/3390093/printer.cfg.txt)

KevinOConnor, 2019-07-29:
I'm going to close this issue as it looks like the original poster found the answer to the original question.

-Kevin

closed 2019-07-29 by KevinOConnor