Klipper3d/klipper #1782: BLTouch offset in screws_tilt_adjust and bed_mesh

wormiedk, 2019-07-03:
I have a prusa clone with the following bltouch section:
[bltouch]
sensor_pin: ar18
control_pin: ar6
pin_move_time: 1
pin_up_reports_not_triggered: True
pin_up_touch_mode_reports_triggered: False
x_offset: 37
y_offset: 2
z_offset: 1.1
speed: 2

My bed is 200x200.

The instructions for bed_mesh states that I should specify the Nozzle position for min and max (see below). 
However if I do that my BLTouch moves outside the bed when trying to probe as the BLtouch is then as 200+37 which is outside the Bed.  I would expect the bed_mesh section to sutract the x an y offsets from the max and min points when probing.

Is the mesh corrected for the offset? Otherwise the mesh will be shifted 37,2 in my case. 

The same applies for the [screws_tilt_adjust] section in which I manually have to subtact the offsets from the probe positions.

#min_point:
#   Defines the minimum x,y position to probe when for rectangular
#   beds. Note that this refers to the nozzle position, take care that
#   you do not define a point that will move the probe off of the bed.
#   This parameter must be provided for rectangular beds.
#max_point:
#   Defines the maximum x,y position to probe when for rectangular
#   beds. Follow the same precautions as listed in min_point. Also note
#   that this does not necessarily define the last point probed, only
#   the maximum coordinate. This parameter must be provided.

wormiedk, 2019-07-03:
`[printer]
kinematics: cartesian
max_velocity: 100
max_accel: 1200 #was 900
max_z_velocity: 20
max_z_accel: 100

[bltouch]
sensor_pin: ar18
control_pin: ar6
pin_move_time: 1
pin_up_reports_not_triggered: True
pin_up_touch_mode_reports_triggered: False
x_offset: 37
y_offset: 2
z_offset: 1.1
speed: 2

[mcu]
serial: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AC01SLRG-if00-port0
pin_map: arduino

[force_move]
enable_force_move: True

[virtual_sdcard]
path: ~/.octoprint/uploads/

[stepper_x]
step_pin: ar54
dir_pin: !ar55
enable_pin: !ar38
step_distance: .0025 #til 32 microsteps (1/800)
endstop_pin: ^ar3
#endstop_pin: ^ar2
position_endstop: 0
position_max: 200
homing_speed: 50

[tmc2208 stepper_x]
uart_pin: ar63
tx_pin: ar59
microsteps: 16
interpolate: false
run_current: 0.6
hold_current: 0.3
stealthchop_threshold: 0

[stepper_y]
step_pin: ar60
dir_pin: !ar61
enable_pin: !ar56
step_distance: .0025
endstop_pin: ^ar14
#endstop_pin: ^ar15
position_endstop: 0
position_max: 200
homing_speed: 50

[tmc2208 stepper_y]
uart_pin: ar40
tx_pin: ar64
microsteps: 16
interpolate: False
run_current: 0.7
hold_current: 0.3
stealthchop_threshold: 0

[stepper_z]
step_pin: ar46
[... 198 lines cut ...]
#*# x_count = 3
#*# y_count = 3
#*# min_x = 0.0
#*# max_x = 160.0
#*# min_y = 30.0
#*# max_y = 160.0
#*# x_offset = 37.0
#*# y_offset = 2.0
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = lagrange
#*# tension = 0.2
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 32.535
#*# pid_ki = 2.582
#*# pid_kd = 102.486
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 74.380
#*# pid_ki = 1.631
#*# pid_kd = 847.935
`

KevinOConnor, 2019-07-10:
In order for us to assist, you need to attach the Klipper log file as described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

wormiedk, 2019-07-11:
I think your tracker is overly focused on log files. This is NOT an error it is merely unexpected behaviour or a poor description in the readme file. As the bed leveling file states that I should note the position based on nozzle position I would assume that the auto bed leveling and screw adjustment procedures would use the nozzle offset. However, I can observe that this is not the case. I am therefore wondering if the generated mesh is correct with regard to the nozzle or if the map is actually a map of bed height at the probe position and therefore inaccurate. 
I fail to see how a log file would help?

KevinOConnor, 2019-07-21:
The code works as it is documented - the position specified in the config are the nozzle positions (the coordinates for the nozzle) - to position the probe at a given location specify the coordinates of where the nozzle would be to obtain the given probe position.   If you've experienced some sort of issue with this, you'll need to include the log.

-Kevin

closed 2019-07-21 by KevinOConnor