Klipper3d/klipper #1783: Homing(G28) result not always same

WindoC, 2019-07-03:
I found abnormal for my printer there is some time save a not good homing result and it makes me sometimes crashing into bed during start a new print.

Remark: I am using optical endstop 670A-WR and the quality should be fine.

So it set a test script as below and repeat it several times. [result.txt](https://github.com/KevinOConnor/klipper/files/3355731/result.txt)

script: [test_probe.sh](https://github.com/KevinOConnor/klipper/files/3355732/test_probe.sh.txt)
```
#!/bin/bash

echo Send command to Klipper
cat >/tmp/printer <<EOF
G28
G1 Z10.00 F6000
PROBE
G1 Z10.00 F6000
PROBE
G1 Z270.00 F6000
M84
EOF

sleep 5
echo Result
cat /tmp/printer &

sleep 9
if [ `ps -ef | grep /tmp/printer | grep -v grep | wc -l` -gt 0 ]
then
	kill `ps -ef | grep /tmp/printer | grep -v grep | awk '{print $2}'`
fi

```

SUM of probe result. The correct z value should be 0.1x but as you see, something it is 1.x and 2.x
```
	Line 7: // probe at 0.891,1.134 is z=0.106266
	Line 11: // probe at 0.891,1.134 is z=0.106266
	Line 20: // probe at 0.891,1.134 is z=2.215641
	Line 24: // probe at 0.891,1.134 is z=2.209391
	Line 33: // probe at 0.891,1.134 is z=1.484391
	Line 37: // probe at 0.891,1.134 is z=1.540641
	Line 46: // probe at 0.891,1.134 is z=0.100016
	Line 50: // probe at 0.891,1.134 is z=0.109391
	Line 59: // probe at 0.891,1.134 is z=0.125016
	Line 63: // probe at 0.891,1.134 is z=0.131266
	Line 72: // probe at 0.891,1.134 is z=0.100016
	Line 76: // probe at 0.891,1.134 is z=0.112516
	Line 85: // probe at 0.891,1.134 is z=0.128141
	Line 89: // probe at 0.891,1.134 is z=0.125016
	Line 98: // probe at 0.891,1.134 is z=2.162516
	Line 102: // probe at 0.891,1.134 is z=2.165641
	Line 111: // probe at 0.891,1.134 is z=0.112516
	Line 115: // probe at 0.891,1.134 is z=0.106266
	Line 124: // probe at 0.891,1.134 is z=0.109391
	Line 128: // probe at 0.891,1.134 is z=0.118766
	Line 137: // probe at 0.891,1.134 is z=0.096891
	Line 141: // probe at 0.891,1.134 is z=0.100016
	Line 150: // probe at 0.891,1.134 is z=0.106266
	Line 154: // probe at 0.891,1.134 is z=0.112516
	Line 163: // probe at 0.891,1.134 is z=0.165641
	Line 167: // probe at 0.891,1.134 is z=0.162516
	Line 176: // probe at 0.891,1.134 is z=1.881266
	Line 180: // probe at 0.891,1.134 is z=1.871891
	Line 189: // probe at 0.891,1.134 is z=2.587516
	Line 193: // probe at 0.891,1.134 is z=2.593766
```

My PROBE quality should be good!
```
0:00:13.539: #PROBE_ACCURACY REPEAT=10 Z=10.0
0:00:13.541: // probe accuracy: at X:0.891 Y:1.134 Z:10.000
0:00:13.542: // and read 10 times with speed of 5 mm/s
0:00:15.881: // probe at 0.891,1.134 is z=0.112516
0:00:24.446: // probe at 0.891,1.134 is z=0.118766 (2)
0:00:28.728: // probe at 0.891,1.134 is z=0.121891
0:00:33.006: // probe at 0.891,1.134 is z=0.115641
0:00:37.284: // probe at 0.891,1.134 is z=0.112516
0:00:41.568: // probe at 0.891,1.134 is z=0.115641
0:00:45.854: // probe at 0.891,1.134 is z=0.121891
0:00:50.137: // probe at 0.891,1.134 is z=0.112516
0:00:54.420: // probe at 0.891,1.134 is z=0.115641
0:00:54.422: // probe accuracy results: maximum 0.121891, minimum 0.112516, average 0.116579, median 0.115641, standard deviation 0.003437
```

[klippy.log](https://github.com/KevinOConnor/klipper/files/3355733/klippy.log)

KevinOConnor, 2019-07-03:
It's not unusual for endstops to have some error in them.  (That is, they may not always trigger at the same time.)

Since you are using tmc2208 drivers, the endstop phase code may improve the accuracy of the endstops - https://www.klipper3d.org/Endstop_Phase.html

If you want to check the repeatability of the endstops, you can run G28 and GET_POSITION multiple times.  Look at the reported "mcu" position - ideally it should be identical after every G28 request.  If it differs, it indicates the endstop is not triggering at the same time.

Finally, you may want to reduce your homing speed.

-Kevin

WindoC, 2019-07-04:
It is not the problem of Klipper. The problem caused by bad installation for the endstop.
I will close this case.

closed 2019-07-04 by WindoC