Klipper3d/klipper #1784: Slice Engineering Thermistor

Shenkoe, 2019-07-03:
This is more of an request, rather than an issue.

My question is, is there a way I could (or someone else) add the the thermistor table for Slice engineering thermistor table to be added into the config or wherever it is supposed to go within Klipper. I have no knowledge as to how to add this thermistor to klipper and have yet to find any information regarding it either. 

I have been in touch with Slice Engineering and Dan barousse ensures me they have not messed around with klipper and have no knowledge as to how to add it in. I reached out to a few prominent folks on youtube as well, and again no one has done this with klipper. 

I know this will take time and it is not something I need anytime soon, but I  would love to be able to know it can be done and how.

Again I would be willing to do this research myself..... but I do not know how to code. I can read it but not make code (weird I know)

This here is the temp table for the thermistor in question > [Slice Thermistor](https://docs.google.com/spreadsheets/d/e/2PACX-1vTHfmSmnGNEuHzjCHEjkA9_DUgnCD9z5E_7Gk1BjUgBYEBtJ6Dlj3hhnFuhFeOfnIc_7zE69VS1tAbx/pubhtml)
This is the thermistor > [50W Heater Cartridge](https://www.sliceengineering.com/shop/50w-heater-cartridge)

KevinOConnor, 2019-07-03:
See the "Custom thermistor" section of config/example-extras.cfg.  There's also an example of a custom thermistor in config/generic-printrboard-g2.cfg.

-Kevin

Shenkoe, 2019-07-03:
Ok, I will take a look at that and see what I can do, if anything. Again im no good at this stuff, I understand it all, I just do not understand enough to effectively write my own code if needed. I will be in touch.

Thanks 
-James aka "SHENKOE"

Shenkoe, 2019-07-03:
@KevinOConnor 
So I was able to look into the config and place the values in the ADC section, but I may be doing it wrong. I have literally messed with klipper once and stopped when I got the thermistor setup portion.

**So this is what I have thus far:**

# Custom ADC temperature sensors (one may define any number of
# sections with an "adc_temperature" prefix). This allows one to
# define a custom temperature sensor that measures a voltage on an
# Analog to Digital Converter (ADC) pin and uses linear interpolation
# between a set of configured temperature/voltage (or
# temperature/resistance) measurements to determine the
# temperature. The resulting sensor can be used as a sensor_type in a
# heater section. (For example, if one defines a "[adc_temperature
# my_sensor]" section then one may use a "sensor_type: my_sensor" when
# defining a heater.) Be sure to place the sensor section in the
# config file above its first use in a heater section.
[adc_temperature my_sensor]
#temperature1:
#voltage1:
#temperature2:
#voltage2:
#...
#   A set of temperatures (in Celsius) and voltages (in Volts) to use
#   as reference when converting a temperature. At least two
#   measurements must be provided.
temperature1:20
resistance1:613.4
temperature2:430
resistance2:0.1805
#...
#   Alternatively one may specify a set of temperatures (in Celsius)
#   and resistance (in Ohms) to use as reference when converting a
#   temperature. At least two measurements must be provided.

Let me know If I placed these values in there correctly. The ADC table is linked in my first post. Again this is all new to me, and I having nothing to really reference other than other issues posted. 

_But again no one is using my exact thermistor, well none that I can tell, so I am kinda winging it with this (**which I do not want to do**)._

cosminr86, 2019-07-04:
User their table from their site 
[image]

Shenkoe, 2019-07-04:
I did use their temp table chart. I'm just unsure if I'm inviting the info in correctly. 

Am I supposed to put the values for the temperatures as I did in my last reply, or should they input differently.  I used the values of the ambient temps and the max I can achieve, referencing the values on the right side of the chart.

cosminr86, 2019-07-04:
The way you did it is fine. Put more values to be more precise.

Shenkoe, 2019-07-05:
@cosminr86 thanks I will add more and try to upload it tonight. I will post back with results and a log if needed. 

Regarding more values, should I do as they did in this post [here](https://github.com/KevinOConnor/klipper/issues/246#issuecomment-374054810), as the section I used only has place for 2 sets of values I used or can I add more to that specific section I posted above. 

Just confusing to me as there are 3 different places to add values to and then uncomment. I think that is my overall issue now, Knowing where to correctly place needed values. 

Sorry for my ignorance or lack of knowledge pertaining to this stuff. But hey I'm trying to learn and understand it at the very least.

- James "SHENKOE" C.

KevinOConnor, 2019-07-05:
You should use a `[thermistor my_thermistor]` type section with three temperature/resistance pairs from the thermistor specification.  See config/generic-printrboard-g2.cfg as an example config that does this.  You should not use an adc_temperature section.

-Kevin

Shenkoe, 2019-07-05:
@KevinOConnor

Ok understood, so more on the lines of something like this 

```
# Custom thermistors (one may define any number of sections with a
# "thermistor" prefix). A custom thermistor may be used in the
# sensor_type field of a heater config section. (For example, if one
# defines a "[thermistor my_thermistor]" section then one may use a
# "sensor_type: my_thermistor" when defining a heater.) Be sure to
# place the thermistor section in the config file above its first use
# in a heater section.
#[thermistor my_thermistor]
#temperature1:20
#resistance1:613.4
#temperature2:200
#resistance2:3.734
#temperature3:430
#resistance3:0.1805
#   Three resistance measurements (in Ohms) at the given temperatures
#   (in Celsius). The three measurements will be used to calculate the
#   Steinhart-Hart coefficients for the thermistor. These parameters
#   must be provided when using Steinhart-Hart to define the
#   thermistor.
```

Sorry I am away from my PC (personal) and cannot test these until later on. But as I mentioned just making sure I am inputting values correctly. 

Thanks for the time you guys have taken to look at my question/ issue. Will post back once I a complete the actual steps, now that I know where to put the info in exactly.

-James AKA "SHENKOE"

fikren23, 2019-07-07:
should point out that klipper is looking for resistance in Ohms, slice's table is listed as kOhm so you'll need to convert those values you listed into Ohms

Shenkoe, 2019-07-08:
@fikren23 thanks for that info. Didn't realize I was using the wrong info, That is another question I had for a later time if issues persisted. I'm going to close this for now as I have other issues I need to work out with my setup, before I can even tackle this portion. 

Thanks to all of you for the assistance
-James AKA "SHENKOE"

closed 2019-07-08 by Shenkoe

shaiss, 2019-07-15:
> @fikren23 thanks for that info. Didn't realize I was using the wrong info, That is another question I had for a later time if issues persisted. I'm going to close this for now as I have other issues [...]

Shenoke, did you figure out the proper slice thermistor table?  I'm about to set mine up

fikren23, 2019-07-15:
I set one up using three sets of values from slice’s Table evenly spaced so I chose 100,200,300 C and the corresponding Ohms values. As I am still building the printer I only did a test to make sure klipper was reading it and did a pid tune on it. Thermistor appears stable above 45ish C but has a +-2 to 4 degree bounce below 45ish C which isn’t necessarily uncommon with a high temp thermistor using those values in a custom thermistor my_thermistor section.

Shenkoe, 2019-07-15:
@shaiss Unfortunately I ran into compiling issues outside the thermistor settings, so I shelved it for the moment and wanted to test it out on my other (stock) ender 3, as I thought it would be best to try it out on something "EASIER" as this was the first time I had tried to set up Klipper in general. 

I plan on trying again later on this week, when I have the time. Now I have to learn how to set up klipper to control multiple MCUs, as I only have the one RPI.... Oh how I love creating more work for myself, never a boring moment. 

@fikren23 I know it's simple, it was just foreign to me at the time. Now that I know what I'm looking at and for I believe I can get it fairly easily. I will need to use a larger range though, as I also have their heater as well which get up to the 430° range (w/ an enclosure) per Slice Engineers emails to me. But now that I have the correct info, regarding this and other things in my setup I should be good to go.

fikren23, 2019-07-15:
The range should have no bearing on the max temp it’s just a range so the Steinhart-Hart coefficients can be calculated, meaning in theory you could use 10,20,30 or 300,400,500