Klipper3d/klipper #1786: os.chmod(filename, 0660) ,SyntaxError: invalid token

3eggert, 2019-07-05:
I get the following error at startup. As far as I see no logfile has been written:

```
octoprint@omv:~/klipper$ /home/octoprint/klippy-env/bin/python /home/octoprint/klipper/klippy/klippy.py /home/octoprint/printer.cfg -l /tmp/klippy.log
Traceback (most recent call last):
  File "/home/octoprint/klipper/klippy/klippy.py", line 8, in <module>
    import util, reactor, queuelogger, msgproto, homing
  File "/home/octoprint/klipper/klippy/util.py", line 39
    os.chmod(filename, 0660)
                          ^
SyntaxError: invalid token
```

I'm trying to install the current version. I tried to identify the problem, but had no luck at all.

KevinOConnor, 2019-07-09:
It appears you've tried to run Klipper with python3 - Klipper requires python2.  More generally, though, if you are going to perform a custom installation of the software, you'll need to be able to track down these types of issues.

-Kevin

closed 2019-07-12 by klipper-gitissuebot

3eggert, 2019-07-14:
Thank you for your support, I was sure to use Python 2.7, but you were right.