Klipper3d/klipper #1787: [FIRMWARE RETRACTION] Z-Hop not taken into acount during z moves

Nitrock28, 2019-07-05:
There seems to be an issue when using firmware retraction with a non-zero Z-hop, where if a move containing a Z coordinate is issued after retraction, the Z hop offset is not taken into account. 

I am using the prusa-slicer on my delta printer and the Gcode looks like this at the beginning : 

```
****Heaters startup*****
G92 E0
G10 ; retract
G92 E0
G1 Z0.300 F9000.000
G1 X-0.004 Y-37.308
G11 ; unretract
G92 E0
****First move****
```
 I have activated the "retraction at layer change" feature, so the same problem happens on every layer. The retraction happens before the new Z value is set and the unrectract after.

What currently happens is that the toolhead actually moves at the requested Z position (0.3mm from the bed) and when the unretract is issued, the nozzle crashes in the bed as the Z-Hop is undone.

Shouldn't the Zhop be treated the same way as a Gcode offset? I am pretty sure Marlin treats it that way.

[klippy.zip](https://github.com/KevinOConnor/klipper/files/3362885/klippy.zip)

KevinOConnor, 2019-07-05:
This is a known issue that was discussed at the end of PR #1617.  Unfortunately, there does not seem to be an easy fix for this.  It may be necessary to remove the z_hop capability from the firmware_retraction module.

-Kevin

KevinOConnor, 2019-07-10:
FYI, I removed support for z_hop (commit 0343d0c7) until we can get this sorted out.

-Kevin

Nitrock28, 2019-07-11:
Ok it makes sense,
I temporarly solved my issue using gcode macros with a variable for the retracted state, and using the gcode_offset to do the Zhop as suggested in PR #1617.
For the most part it works, but I had to take make a "safe" set_gcode_offset alias macro to take the curent value of Z_Hop into account, otherwise tuning the first layer can be dangerous if the offset is applied during a retraction. I don't have the macros on hand at the moment but I can post them later if it's of any interest.

KevinOConnor, 2019-07-11:
FYI, in a separate conversation I had with @Arksine, we came up with two possible solutions:

- Record the current g-code z-position at the start of G10 (via `gcode.get_status()['gcode_position'].z`) and only restore the z position on a G11 if the current g-code z-position is where it should be (`g11_z_position == g10_z_position + z_hop`).  This scheme doesn't work well if the print is in relative mode, but that could be worked around by being more lenient if G11 is called while in relative mode (`g11_z_position >= g10_z_position + z_hop`).  This scheme doesn't work well if the user's z_hop is set exactly the same as the print's layer height (because `g11_z_position == g10_z_position + z_hop` would result in a false-positive on a layer change), but that could be worked around by adding 0.000000007 to the user's configured z_hop (thus making it exceedingly unlikely the layer height is the same as the z_hop).

- At the start of the G10, register new G1 and G0 command handlers.  These overridden handlers would detect any z movement requests and disable z_hop in that case.  On a G11 the original handlers would be restored.

-Kevin

Nitrock28, 2019-07-11:
Ok, I think I understand what you want to do. I didn't understand at first why remove the ZHop in case of Z movement, but you simply let the "UnZhop" be handled by the normal gcode. That would be a solution I think.

As I said I implemented a few macros to mimic what other slicers do (Marlin at least) which is to add a separate offset in z during the retraction, so the Z movements are handled naturally and the offset is removed during the unretract.
Here is what I have done in my gcode_macros 

```
[gcode_macro SET_RETRACTION]
default_parameter_RETRACT_LENGTH = 4
default_parameter_RETRACT_SPEED = 6000
default_parameter_UNRETRACT_EXTRA_LENGTH = 0
default_parameter_Z_HOP = 2
default_parameter_Z_HOP_SPEED = 3000
variable_rl: 4
variable_rs: 6000
variable_uel: 0
variable_zh: 2
variable_zhs: 3000
gcode:
	SET_GCODE_VARIABLE MACRO=SET_RETRACTION VARIABLE=rl VALUE={RETRACT_LENGTH}
	SET_GCODE_VARIABLE MACRO=SET_RETRACTION VARIABLE=rs VALUE={RETRACT_SPEED}
	SET_GCODE_VARIABLE MACRO=SET_RETRACTION VARIABLE=uel VALUE={UNRETRACT_EXTRA_LENGTH}
	SET_GCODE_VARIABLE MACRO=SET_RETRACTION VARIABLE=zh VALUE={ Z_HOP }
	SET_GCODE_VARIABLE MACRO=SET_RETRACTION VARIABLE=zhs VALUE={Z_HOP_SPEED}
	M118 Retraction modified :  {RETRACT_LENGTH} | {RETRACT_SPEED} | {UNRETRACT_EXTRA_LENGTH} | {Z_HOP} | {Z_HOP_SPEED}

[gcode_macro G10]
#Retraction
variable_retracted: 0
gcode:
	{% if not printer["gcode_macro G10"].retracted %}
		{% if printer.gcode.abs_extrude %}
			SET_GCODE_VARIABLE MACRO=G11 VARIABLE=abs_e VALUE=1
			M83
		{% else %}
			SET_GCODE_VARIABLE MACRO=G11 VARIABLE=abs_e VALUE=0
		{% endif %}
		G1 E-{printer["gcode_macro SET_RETRACTION"].rl} F{printer["gcode_macro SET_RETRACTION"].rs}
		SET_GCODE_OFFSET Z_ADJUST={printer["gcode_macro SET_RETRACTION"].zh} MOVE_SPEED={printer["gcode_macro SET_RETRACTION"].zhs} MOVE=1
		{% if printer["gcode_macro G11"].abs_e %}
			M82
		{% endif %}
		SET_GCODE_VARIABLE MACRO=G10 VARIABLE=retracted VALUE=1
	{% else %}
		M118 Already retracted
	{% endif %}
	
[gcode_macro G11]
#Unretraction
variable_abs_e: 0
gcode:
	{% if printer["gcode_macro G10"].retracted %}
		{% if printer.gcode.abs_extrude %}
			SET_GCODE_VARIABLE MACRO=G11 VARIABLE=abs_e VALUE=1
			M83
		{% else %}
			SET_GCODE_VARIABLE MACRO=G11 VARIABLE=abs_e VALUE=0
		{% endif %}
		SET_GCODE_OFFSET Z_ADJUST=-{printer["gcode_macro SET_RETRACTION"].zh} MOVE_SPEED={printer["gcode_macro SET_RETRACTION"].zhs} MOVE=1
		G1 E{printer["gcode_macro SET_RETRACTION"].rl+printer["gcode_macro SET_RETRACTION"].uel} F{printer["gcode_macro SET_RETRACTION"].rs}
		{% if printer["gcode_macro G11"].abs_e %}
			M82
		{% endif %}
		SET_GCODE_VARIABLE MACRO=G10 VARIABLE=retracted VALUE=0
	{% else %}
		M118 Already unretracted
	{% endif %}

[gcode_macro SAFE_OFFSET]
# alias to SET_GCODE_OFFSET taking the retraction state into account, used to tune the first layer Z offset
default_parameter_ZL : 0
default_parameter_MOVEL : 0
gcode:
	{% if printer["gcode_macro G10"].retracted %}
		SET_GCODE_OFFSET Z={printer["gcode_macro SET_RETRACTION"].zh|float + ZL|float} MOVE={MOVEL}
	{% else %}
		SET_GCODE_OFFSET Z={ZL} MOVE={MOVEL}
	{% endif %}
```

I also reset the retraction state and gcode_offset in the homing sequence. My printer is quite simple so I may miss a few cases in which this method would not work.

KevinOConnor, 2019-07-11:
FYI, there's two failure cases with the previous z_hop code - a Z layer change between G10 and G11, and a G10 at the end of a print followed by a G11 at the start of the next print.

It's possible to solve the first failure case by using SET_GCODE_OFFSET (or via a similar mechanism using G92), but as far as I know it just complicates the second failure case.

-Kevin

KevinOConnor, 2019-08-17:
I'm going to mark this as inactive as it doesn't look like it is currently being implemented.

-Kevin

closed 2019-08-17 by KevinOConnor