Klipper3d/klipper #1790: Not an issue - more of a question

dushyantahuja, 2019-07-06:
Was just wondering if the endstops use interrupts, or are polled. My Z homing is inconsistent - nothing major, just a few fractions of mm here and there - but need to re-adjust the bed every time. Was wondering if I need to redesign my Z. I have a belt based Z axis with step_distance: .0003125

Hywelmartin, 2019-07-06:
try [[endstop_phase]](https://github.com/KevinOConnor/klipper/blob/0f1d42466b60ab37c8c7ec44b79835327cb4375b/config/example-extras.cfg#L509)

KevinOConnor, 2019-07-07:
Endstops are polled, but polled at a high enough frequency that irqs would not improve anything.  For any further assistance, you'll need to attach the Klipper log file.

-Kevin

dushyantahuja, 2019-07-07:
Great, thank you so much.
I was just wondering since my mm per step is so small - maybe the polling
wasn't working as designed. Anyways - I've decided to redesign my Z - for
other reasons. I'll close this for now.
------------------
Dushyant Ahuja

closed 2019-07-07 by dushyantahuja