Klipper3d/klipper #1793: Temperature jumps massively when heating

lights0123, 2019-07-07:
Sometimes, when heating, the extruder thermistor temperature jumps a massive amount, either upwards or downwards. For example, in this case, the temperature jumped up by 75°C instantly, triggering an emergency shutdown. 
[image]

Additionally, sometimes during printing, the temperature falls a large amount, also triggering an emergency shutdown. I don't have a log of that now, but if it happens again, I will post an update. It seems to be a "fake" fall, as klipper responds by increasing the temperature rapidly, but this causes a massive increase after a few seconds.

I would assume faulty wiring—however, the same setup worked just fine on the stock [Repetier-ish firmware](https://github.com/seemecnc/Firmware). The same thing happens when the stepper motors are disabled. I am using an E3D v6 Chimera dual-extruder setup on a Rostock Max V2, using the stock RAMBo board.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3365849/klippy.log)

lights0123, 2019-07-07:
And during a print:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3366109/klippy.log)

KevinOConnor, 2019-07-10:
I suspect you are overloading your power supply (it may fail to maintain a stable voltage when the heaters are on).  You may wish to try reducing the `max_power` settings (eg, set max_power=0.3 in heater_bed, and max_power=0.8 in the extruders).

-Kevin

closed 2019-07-29 by KevinOConnor