Klipper3d/klipper #1794: Enhanced delta calibration - the correct position of the calibration object

mbgroot, 2019-07-07:
How should the calibration object be positioned correctly in relation to the towers? I use the Kura. When printed, it turns out that the letter C is printed near the tower X, the letter A near the tower Y, the letter B near the tower Z.
I need to turn the figure in the slicer, what would be A-X, B-Y, C-Z? Or not?

BlackStump, 2019-07-07:
The location of your steppers should be X, Y, Z in a counterclockwise direction and if you look here 
https://github.com/KevinOConnor/klipper/blob/master/config/example-delta.cfg#L10
and here
https://github.com/KevinOConnor/klipper/blob/master/config/example-delta.cfg#L36
and here
https://github.com/KevinOConnor/klipper/blob/master/config/example-delta.cfg#L45

it tells you where your steppers should be if that is correct then you should not have to orientate in a slicer just slice and print

mbgroot, 2019-07-08:
> it tells you where your steppers should be if that is correct

Thank you! Thank you! I have to turn the figure in the slicer...

closed 2019-07-08 by mbgroot