Klipper3d/klipper #1795: SKR 1.3 not working

tonn0297, 2019-07-08:
Hello. 
i have a skr v1.3 and i can´t load the firmware over. i have tried both via the installer and taking the .bin file and putting it on the sd card. if i put it on the sd card it just stalls. like its loading firmware but never gets done. i have selected the right prossesor.

i hope you can help me.

gloomyandy, 2019-07-08:
Probably doesn't help but I have a version of Klipper from 7 days ago working fine on an SKR V3.1. Initial upload was via the SD card which worked fine, after that (with klipper firmware running) I was able to use make flash <usb device> to upload changes. 

Probably worth trying a different SD card and/or reformatting the card. Perhaps a silly question but how did you get the bin file onto the SD card? Could the file have been corrupted in the process of moving it?

Here is my .config file if it helps...
https://github.com/gloomyandy/klipper/blob/myskr1.3/.config

tonn0297, 2019-07-09:
i winscp on the the pc and took it out after i did the make

tonn0297, 2019-07-09:
and mine config looks like this= 
/*
 *
 * Automatically generated file; DO NOT EDIT.
 * Klipper Firmware Configuration
 *
 */
#define CONFIG_STM_FLASH_START_7000 0
#define CONFIG_HAVE_SERCOM 0
#define CONFIG_MACH_SAMD51N19 0
#define CONFIG_SERIAL_BAUD 250000
#define CONFIG_MACH_LPC1768 1
#define CONFIG_AVR_FREQ_16000000 0
#define CONFIG_MACH_ATSAMD 0
#define CONFIG_MACH_SAMD21E15 0
#define CONFIG_FLASH_SIZE 0
#define CONFIG_MACH_SAMD51 0
#define CONFIG_SERIAL_BAUD_U2X 0
#define CONFIG_MACH_atmega328 0
#define CONFIG_UNAME_RELEASE "5.0.0-20-generic"
#define CONFIG_STM_FLASH_START_0000 0
#define CONFIG_MACH_LINUX 0
#define CONFIG_MACH_atmega644p 0
#define CONFIG_MACH_SAMD51G19 0
#define CONFIG_STEP_DELAY 2
#define CONFIG_ATSAMD_SELECT 0
#define CONFIG_BOARD_DIRECTORY "lpc176x"
#define CONFIG_MACH_SAM3X8C 0
#define CONFIG_USE_LOG ""
#define CONFIG_DEBUG_OUT 0
#define CONFIG_INITIAL_PINS ""
#define CONFIG_AVR_SERIAL_UART0 0
#define CONFIG_CLEAR_PRESCALER 0
#define CONFIG_SIMULATOR_SELECT 0
#define CONFIG_USBSERIAL 0
#define CONFIG_MACH_SAM4E 0
#define CONFIG_AVR_CLKPR 
#define CONFIG_STACK_SIZE 
#define CONFIG_MACH_LPC1769 0
#define CONFIG_MACH_LPC176X 1
#define CONFIG_SERIAL_PORT 
#define CONFIG_MACH_at90usb1286 0
#define CONFIG_ATSAM_SELECT 0
#define CONFIG_HAVE_GPIO_HARD_PWM 0
#define CONFIG_MACH_SAM4 0
#define CONFIG_MACH_SAM3X8E 0
#define CONFIG_MACH_STM32F0 0
#define CONFIG_HAVE_GPIO 1
#define CONFIG_AVR_FREQ_20000000 0
#define CONFIG_MACH_SAM4S8C 0
#define CONFIG_LINUX_SELECT 0
#define CONFIG_FLASH_START_2000 0
#define CONFIG_AVRDUDE_PROTOCOL ""
#define CONFIG_STM_FLASH_START_2000 0
#define CONFIG_FLASH_START_0000 0
#define CONFIG_MACH_atmega328p 0
#define CONFIG_LOW_LEVEL_OPTIONS 1
#define CONFIG_USB_SERIAL_NUMBER ""
#define CONFIG_MACH_SAM4E8E 0
#define CONFIG_FLASH_START 0x0000
#define CONFIG_RAM_SIZE 0
#define CONFIG_MACH_SAMD21E18 0
#define CONFIG_MACH_SAMD21 0
#define CONFIG_HAVE_GPIO_ADC 1
#define CONFIG_MACH_SAMD21G18 0
#define CONFIG_SIMULAVR 0
#define CONFIG_CLOCK_REF_X32K 0
#define CONFIG_AVR_SERIAL_UART1 0
#define CONFIG_MACH_AVR 0
#define CONFIG_MACH_ATSAM 0
#define CONFIG_MACH_atmega32u4 0
#define CONFIG_HAVE_GPIO_I2C 1
#define CONFIG_MACH_SAMD51J19 0
#define CONFIG_MACH_SAMD51P20 0
#define CONFIG_SERIAL 1
#define CONFIG_CLOCK_REF_INTERNAL 0
#define CONFIG_MACH_atmega1280 0
#define CONFIG_AVR_WATCHDOG 0
#define CONFIG_AVR_SERIAL_UART2 0
#define CONFIG_MACH_SIMU 0
#define CONFIG_MCU ""
#define CONFIG_MACH_STM32F1 0
#define CONFIG_CLOCK_FREQ 100000000
#define CONFIG_STM32F1_SELECT 0
#define CONFIG_USB_DEVICE_ID 0
#define CONFIG_SMOOTHIEWARE_BOOTLOADER 0
#define CONFIG_PRU_SELECT 0
#define CONFIG_MACH_at90usb646 0
#define CONFIG_HAVE_GPIO_SPI 1
#define CONFIG_MACH_SAM4S 0
#define CONFIG_USB_VENDOR_ID 0
#define CONFIG_CUSTOM_STEP_DELAY 0
#define CONFIG_FLASH_START_4000 0
#define CONFIG_HAVE_GPIO_BITBANGING 1
#define CONFIG_LPC_SELECT 1
#define CONFIG_MACH_atmega2560 0
#define CONFIG_STM32F0_SELECT 0
#define CONFIG_AVR_SELECT 0
#define CONFIG_CANSERIAL 0
#define CONFIG_AVR_FREQ_8000000 0
#define CONFIG_MACH_atmega168 0
#define CONFIG_AVR_STACK_SIZE 
#define CONFIG_MACH_PRU 0
#define CONFIG_INLINE_STEPPER_HACK 1
#define CONFIG_AVR_SERIAL_UART3 0
#define CONFIG_MACH_SAM3X 0
#define CONFIG_MACH_atmega1284p 0

grblmpfh, 2019-07-11:
Did you rename the "klipper.bin" to "firmware.bin" on the sd-card?

closed 2019-07-15 by klipper-gitissuebot

elliotf, 2019-07-15:
@tonn0297 I had issues when I accidentally said "no" to the smoothie bootloader, which is looks like you have done:

```#define CONFIG_SMOOTHIEWARE_BOOTLOADER 0```