Klipper3d/klipper #1796: CR10 + bltouch configuration

WNijhof, 2019-07-08:
Hello,

I hope someone can help me out here.
I am running klipper for a long time now, but Im in the middle of installing an BLTouch on my printer.
This evening I am trying to get it to work in the code, but it seems I cant get the pins right. Hopefully somone can see my mistake.

I added this code underneath in the printer.conf file.
My BLTouch is connected to the z-stop (Pin 20) and the Buzzer pin (Pin 27)

`[bltouch]
sensor_pin: 20
control_pin: 27
pin_move_time: 0.400
x_offset: 36
y_offset: 0
z_offset: 1.92

[homing_override]
set_position_z: 0
axes: z
gcode:
    G90 ; Use absolute position mode
    G1 Z10 ; Move up 10mm
    G28 X Y
    G1 X114 Y150 F3600
    G28 Z
    G1 Z10
    G1 X0 Y0 F3600

[bed_mesh]
speed: 150
horizontal_move_z: 5
samples: 1
sample_retract_dist: 2.0
min_point: 0, 30
max_point: 233, 270
probe_count: 7,7
fade_start: 1.0
fade_end: 10.0`

[klippy.log](https://github.com/KevinOConnor/klipper/files/3369887/klippy.log)

BlackStump, 2019-07-08:
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-convert-a-marlin-pin-number-to-a-klipper-pin-name
doing a search on closed items about bltouch and creality printers 
your pin numbers are not right
sensor_pin: ^PC4
control_pin: PA4

KevinOConnor, 2019-07-21:
I'm going to close this issue as it looks like the original question was answered as best we could (specify the correct pin names).

-Kevin

closed 2019-07-21 by KevinOConnor