Klipper3d/klipper #1797: TMC5160 suddenly shuts down during the print, sometimes not extruding after restart

andrico21, 2019-07-10:
After latest update from GH my TMC5160 extruder began to fail during the print - mostly on the first layer after some successful extruding.
Also I've noticed sometimes the extruder not working right after Klipper/controller restart, but after manual initialization begin to work (but fail soon in the print process)
[NotExtrudingUntilManualInit_klippy.log](https://github.com/KevinOConnor/klipper/files/3376240/NotExtrudingUntilManualInit_klippy.log)

Stephan3, 2019-07-10:
@andrico21 
First point is that your stealthchop is very low on all motors. Theese drivers arent like 2130. You easily can set that to 200 on a corexy machine. Special mechanics ar impaction that in many ways.

So as a proposal try to use your extruder in spreadcycle first.

andrico21, 2019-07-10:
@Stephan3 thanks, I will try. I'm confused by the fact it worked great before latest update. No single successful print after update - sometimes extruder fail, sometimes leveling is not working

andrico21, 2019-07-10:
Okay, I digged into 'git reflog --date=iso | grep pull' output: previous update I performed at 2019-06-23 16:31 +0100. So the potential "problem" commit was between 2019-06-23 and 2019-07-09.

andrico21, 2019-07-10:
Can confirm now - extruder is not shutting down after resetting to https://github.com/KevinOConnor/klipper/commit/c3467f5b3541c72bc771b1408e9ebf9e8f8bdfff commit

Oranginator, 2019-07-15:
I got the same problem sometimes a motor switches.

KevinOConnor, 2019-07-21:
There weren't any significant changes to the TMC code between c3467f5 and 0f1d424.  My best guess is that the driver didn't receive some of the initi messages - possibly due to a motor power on race, or due to noise on the spi line.  You might want to try sending DUMP_TMC commands to inspect the state of the driver after power up and after the failure.

-Kevin

andrico21, 2019-07-22:
@KevinOConnor please check my log in the thread first message please - I've dumped TMC registers several times there and after extruder fail I tried to init TMCs manually and then the extruder worked fine - TMC dump after is also there.

Oranginator, 2019-07-22:
I'm using 0.9° steppers and I lowered the microsteps from 16 to 8 now they're also working as intended...

andrico21, 2019-07-23:
@Oranginator I'm also using 0.9° for extruder, but bandwidth is enough

KevinOConnor, 2019-07-29:
Alas, I don't have any insight to the cause of this issue.  The best I can offer is that if you (or someone else) can diagnose the root cause of the issue, then I may be able to help with a solution.

If you can obtain and annotate a DUMP_TMC before and after a failure, that may help narrow down the issue.

-Kevin

lekv, 2019-08-05:
@andrico21 Have a look at #1774. It outlines how to debug this issue and it's root cause might be relevant here, too.

andrico21, 2019-08-06:
@lekv thanks mate, I'll check GCONF state in next 2-3 days.

KevinOConnor, 2019-08-17:
Any further updates on this?

-Kevin

Oranginator, 2019-08-17:
I don't know how but mine is working fine atm.

KevinOConnor, 2019-09-06:
I'm going to close this issue as it appears the conversation has concluded.

-Kevin

closed 2019-09-06 by KevinOConnor