Klipper3d/klipper #1798: QUAD_GANTRY_LEVEL is not working as expected after last update

andrico21, 2019-07-10:
It worked before last update, now it's broken.

For example, running first quad_gantry_level, got:
probe at 25.000,-0.000 is z=2.789255
probe at 25.000,-0.000 is z=2.794255
probe at 25.000,-0.000 is z=2.844255
probe at 25.000,-0.000 is z=2.806755
probe at 25.000,200.000 is z=2.666755
probe at 25.000,200.000 is z=2.651755
probe at 25.000,200.000 is z=2.649255
probe at 25.000,200.000 is z=2.659255
probe at 225.000,200.000 is z=2.621755
probe at 225.000,200.000 is z=2.621755
probe at 225.000,200.000 is z=2.621755
probe at 225.000,200.000 is z=2.621755
probe at 225.000,-0.000 is z=2.544255
probe at 225.000,-0.000 is z=2.544255
probe at 225.000,-0.000 is z=2.581755
probe at 225.000,-0.000 is z=2.549255
Gantry-relative probe points:
0: 7.191370 1: 7.343245 2: 7.378245 3: 7.445120
Actuator Positions:
z: 7.038577 z1: 7.446049 z2: 7.318917 z3: 7.571320
Average: 7.343716
Making the following Z adjustments:
stepper_z = 0.305139
stepper_z1 = -0.102333
stepper_z2 = 0.024799
stepper_z3 = -0.227604

next one failed:
probe at 25.000,-0.000 is z=2.705104
probe at 25.000,-0.000 is z=3.945104
**11:29:04 Probe samples exceed samples_tolerance**

next try might work, might not - looks like one of four Z drives moving incorrectly or incorrect probe measurement...
[QuadGantryLevelBroken_klippy.log](https://github.com/KevinOConnor/klipper/files/3377058/QuadGantryLevelBroken_klippy.log)

andrico21, 2019-07-10:
My Z are four standalone TMC 2208 with standard 1.8 steppers.

Stephan3, 2019-07-10:
See this commit:
https://github.com/KevinOConnor/klipper/commit/c92aca74b5a874e4ce274e2979ffaf6ecfe97eb3#diff-2bc03e0a9e80c35c034202dd38860b6c

Your probe accuracy looks a bit off to me in that second try. moar than a mm between the samples isn´t very good.
Anyways you can set `samples_tolerance` in your probe config that fits your needs.

andrico21, 2019-07-10:
@Stephan3 I'm pretty sure it's not about probe accuracy, each time it fails - it fails exactly on the second try on the front left stepper and each time the difference is about 1.2-1.24mm

andrico21, 2019-07-10:
@Stephan3 and see please first try - it was successfull and the second try were made without motors off between. So maybe Z1 is skipping steps while probing while Z2-Z4 are ok or it's something with probe quering

andrico21, 2019-07-10:
Now trying one more time:
Recv: // probe at 25.000,-0.000 is z=2.417500
Recv: // probe at 25.000,-0.000 is z=2.422500
Recv: // probe at 25.000,-0.000 is z=3.657500 - **first abnormal measurement**
Recv: !! Probe samples exceed samples_tolerance
all subsequent measurements in this point now are larger, that's why I suspect Z1 motion issues...
Recv: // probe at 25.000,-0.000 is z=3.530000
Recv: // probe at 25.000,-0.000 is z=3.537500
Recv: // probe at 25.000,-0.000 is z=3.540000
Recv: // probe at 25.000,-0.000 is z=3.542500
Recv: // probe at 25.000,200.000 is z=2.717500
Recv: // probe at 25.000,200.000 is z=2.720000
Recv: // probe at 25.000,200.000 is z=2.702500
Recv: // probe at 25.000,200.000 is z=2.707500
Recv: // probe at 225.000,200.000 is z=2.410000
Recv: // probe at 225.000,200.000 is z=2.405000
Recv: // probe at 225.000,200.000 is z=2.405000
Recv: // probe at 225.000,200.000 is z=2.405000
Recv: // probe at 225.000,-0.000 is z=2.715000
Recv: // probe at 225.000,-0.000 is z=2.725000
Recv: // probe at 225.000,-0.000 is z=2.727500
Recv: // probe at 225.000,-0.000 is z=2.730000
Recv: // Gantry-relative probe points:
Recv: // 0: 6.462500 1: 7.288125 2: 7.593750 3: 7.275625
Recv: // Actuator Positions:
Recv: // z: 5.932425 z1: 7.651567 z2: 7.770684 z3: 7.582455
Recv: // Average: 7.234283
Recv: // Making the following Z adjustments:
Recv: // stepper_z = 1.301857 
Recv: // stepper_z1 = -0.417284
Recv: // stepper_z2 = -0.536402
Recv: // stepper_z3 = -0.348172

Stephan3, 2019-07-10:
What Probe is this? What Printer is this?

andrico21, 2019-07-10:
@Stephan3 Voron2.1, didn't you recognize me dude? :)

andrico21, 2019-07-10:
Okay, I digged into 'git reflog --date=iso | grep pull' output: previous update I performed at 2019-06-23 16:31 +0100. So the potential "problem" commit was between 2019-06-23 and 2019-07-09.

Stephan3, 2019-07-10:
Okay, thought that but wanted to be sure. With the M3 extruder humans experience different things in stealthchop. For me, with tuning it works. For a M3 your current feels high. Probably not related to your issue but try .4 or .6

andrico21, 2019-07-10:
Ok, current status: I had hard reset to https://github.com/KevinOConnor/klipper/commit/c3467f5b3541c72bc771b1408e9ebf9e8f8bdfff - leveling is MUCH MORE stable now: ran 5 or 6 QUAD_GANTRY_LEVEL in series and got no fails, on https://github.com/KevinOConnor/klipper/commit/0f1d42466b60ab37c8c7ec44b79835327cb4375b in most cases second try is failing.
But eventually I still get the same issue even on https://github.com/KevinOConnor/klipper/commit/c3467f5b3541c72bc771b1408e9ebf9e8f8bdfff - but significantly less times...

andrico21, 2019-07-10:
Found mechanical problem - since I cannot explain why the issue occuring more often with last Klipper update than with previously installed I'd like to hold this issue until I fix mechanical problem.

andrico21, 2019-07-12:
Fixed one slipping belt and issue is gone on https://github.com/KevinOConnor/klipper/commit/c3467f5b3541c72bc771b1408e9ebf9e8f8bdfff
not tried yet on https://github.com/KevinOConnor/klipper/commit/0f1d42466b60ab37c8c7ec44b79835327cb4375b but believe this particular issue is solved and not caused by software.

closed 2019-07-12 by andrico21