Klipper3d/klipper #1801: Generate a single pulse on digital output

ardyesp, 2019-07-11:
My board has stepper drivers reset pin connected to MCU and need a reset pulse (low + wait + high) at powerup for reliable driver operation. I tried:
**printer.cfg**
```
[toggle_pin stepper_reset]
pin: ar41
```
and created a module **toggle_pin.py**
```
import time
class PrinterToggleDigitalOut:
    def __init__(self, config):
        printer = config.get_printer()
        ppins = printer.lookup_object('pins')
        mcu_pin = ppins.setup_pin('digital_out', config.get('pin'))
        mcu_pin.setup_start_value(0, 0, True)
        time.sleep(0.1)
        mcu_pin.setup_start_value(1, 1, True)

def load_config_prefix(config):
    return PrinterToggleDigitalOut(config)
```
I don't think this will work, since the second *setup_start_value* will overwrite the first one, when klippy connects to MCU. What would be the right way to implement this? 
Interim, I have to manually issue SET_PIN commands before trying to jog any axis.

KevinOConnor, 2019-07-11:
I suspect the easiest solution would be to define a [homing_override] section and add something like `SET_PIN PIN=stepper_reset_pin VALUE=0 ; G4 P100 ; SET_PIN PIN=stepper_reset_pin VALUE=1 ; G28`

In order to accomplish what you want from an extras module, you'd probably want to define a "klippy:connect" event callback and then perform the `mcu_pin.set_digital()` calls at that point.  There's some details at: https://www.klipper3d.org/Code_Overview.html

-Kevin

ardyesp, 2019-07-12:
Thanks @KevinOConnor, I tried setting up a callback for connect, and see a digital output pulse, but get a "_MCU 'mcu' shutdown: Missed scheduling of next pin event_" 

Interestingly, this only happens, if the "startupValue" is 0. Log file is attached.
```
import logging

class PrinterToggleDigitalOut:
    def __init__(self, config):
        self.printer = config.get_printer()
        self.printer.register_event_handler("klippy:connect", self.handle_connect)
        ppins = self.printer.lookup_object('pins')
        self.mcu_pin = ppins.setup_pin('digital_out', config.get('pin'))
        self.startupValue = config.getint('startup_value', 0)
        self.pulseDelay = config.getfloat('delay', 0.1)
        self.mcu_pin.setup_start_value(self.startupValue, self.startupValue, False)
    def handle_connect(self):
        print_time = self.printer.lookup_object('toolhead').get_last_move_time()
        self.mcu_pin.set_digital(print_time, 1)
def load_config_prefix(config):
    return PrinterToggleDigitalOut(config)
```

[klippy.log](https://github.com/KevinOConnor/klipper/files/3388105/klippy.log)

KevinOConnor, 2019-07-18:
It's not valid to call toolhead.get_last_move_time() from a connect callback.  You'd need to obtain the time from the mcu object (and add a 100ms transmission delay to it).  You can look at how klippy/extras/bltouch.py does this.

-Kevin

ardyesp, 2019-07-18:
Thanks, I am relatively new to python, and lack of data types is making it really hard to understand what is going on in the bltouch code. 

Even with a simple statement like ```self.mcu_pin.set_digital(20, 1)```, mcu shuts down after 20 seconds of being idle, with "missed scheduling" message.

KevinOConnor, 2019-07-21:
Be sure to call `self.mcu_pin.setup_max_duration(0.)` during the init.  There's a little bit of documentation on this in docs/MCU_Commands.md.

-Kevin

ardyesp, 2019-07-21:
Thanks @KevinOConnor. That solved the issue. Pasting the code here in case someone else needs it
Configuration:
```
[toggle_pin stepper_reset]
pin: ar41
startup_value: 0
delay: 0.1
```
Module toggle_pin.py:
```
# Toggle digital output pins after a delay
MIN_CMD_TIME = 0.1

class PrinterToggleDigitalOut:
  def __init__(self, config):
    self.printer = config.get_printer()
    self.printer.register_event_handler("klippy:connect", self.handle_connect)
    togglePin = config.get('pin')
    self.startupValue = config.getint('startup_value', 0)
    self.pulseDelay = config.getfloat('delay', 0.1)
    ppins = self.printer.lookup_object('pins')
    self.mcu_pin = ppins.setup_pin('digital_out', togglePin)
    self.mcu_pin.setup_max_duration(0.)
    self.mcu_pin.setup_start_value(self.startupValue, self.startupValue, False)
  def handle_connect(self):
    curtime = self.printer.get_reactor().monotonic()
    est_time = self.mcu_pin.get_mcu().estimated_print_time(curtime)
    next_cmd_time = est_time + MIN_CMD_TIME + self.pulseDelay
    self.mcu_pin.set_digital(next_cmd_time, (not self.startupValue))
def load_config_prefix(config):
  return PrinterToggleDigitalOut(config)
```

closed 2019-07-21 by ardyesp