Klipper3d/klipper #1802: Strange x-axis deffect when printing from Cura !

PappaBoom, 2019-07-11:
Hi !
Hope some one here might solve this ! Don't know if it is a G-code parsing error or other. I know all is gonna jump and say missed stepps, to tight belts , to lose belts, to fast print speed but i have tested all that and still the same error. I have a big IDEX printer that i built from scratch. Just upgraded it to Klipper from repetiere firmware. It worked like a charm mechanically before the switch. But after the switch all prints lean to the right in X axis.
Im in the phase of tuning all in so i don't go fast. Before it was god at 100mm/s print speeds. Now i print at 50mm/s whit acceleration tuned down to 500. I also tried at 20mm/s print speed and it came out the same way. And the funny part is it looks different depending on what version of cura i use !!!??? So pleas help. Im including pictures and the G-code that cura produced. The wobbly version is cura 3.5.1 and the "straight" version is cura4.0. There are no artefacts in the Y-axis. 

Cheers! 

[Gcode.zip](https://github.com/KevinOConnor/klipper/files/3382470/Gcode.zip)
[klipperconfig.txt](https://github.com/KevinOConnor/klipper/files/3382471/klipperconfig.txt)
[image]
[image]

PappaBoom, 2019-07-15:
Is this the right forum ? Or should i trie somewhere else ?

Im banging my head about this !

So i went over belts and extended the step pulse since I'm using external stepper drivers TB6600. Used 10us on Repetier so i went with the same. Extended it to 15us and no difference. So I'm thinking this is something that Cura puts in the code ! So i sliced with repeater host and the Cura version that came with it ! And presto . A perfect square ! Am i the only person to experience this ?
[image]
Cube to the left Command line cura form repeater host. Cube to the right printed with cura 4.0.0
Cura 3.5 prints even uglier.

Anyway this is not an mechanical error. 

Pleas advise ! Need my printer upp and running and i like Klipper so much more than repetier !!!!!

Attached the Functional G-code that repeater host created!
[CuraRepetierHost_Box202020.zip](https://github.com/KevinOConnor/klipper/files/3392743/CuraRepetierHost_Box202020.zip)

Note ! I print using Octoprint directly from Cura. Multi machine setup. Cura 3.5 are on OSX and 4.0.0 is on PC
//F

BlackStump, 2019-07-15:
maybe you will get more response if you actually read and follow the requirement.
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

KevinOConnor, 2019-08-05:
I'm going to close this issue as it looks like the conversation has concluded.  It appears from the report that cura is sending bad commands to Klipper and it does not appear anyone here knows why cura would do that.

-Kevin

closed 2019-08-05 by KevinOConnor