Klipper3d/klipper #1803: BED_TILT_CALIBRATE doesn't take probe offset into account.

rmawatson, 2019-07-11:
Using the sample point setup for bed tilt, it appears to try and probe off the side of the bed, and does not take into account the x/y offset that has been configured for the [bltouch].

Arksine, 2019-07-16:
The points you specify refer to nozzle position in [bed_tilt] and [bed_mesh] refer to nozzle position.  The mesh itself uses the offsets in the [probe] section to correctly map the mesh to the bed.

rmawatson, 2019-07-16:
right, thanks. good to know the actual mesh is offset by the probes offsets. 

I do however think that the points should be the actual points that are probed, rather than the points being with respect to the nozzle. You otherwise have to manually calculate these point positions to take account of the probe offset get the probe to hit where you want. 

Perhaps an option for this? points_relative_to_probe=True/False

KevinOConnor, 2019-08-05:
I'm going to close this issue as it appears the original question was answered (bed mesh points are declared relative to the nozzle and not the probe).

-Kevin

closed 2019-08-05 by KevinOConnor