Klipper3d/klipper #1804: Z-Homing fails when Z end-stop already pressed

urish, 2019-07-11:
Whenever I home the Z axis and the Z end-stop is already pressed, klipper still tries moving the Z-axis down aggressively. 

It homes correctly if I start with the Z axis a bit up (so that the end-stop isn't pressed), but it failed miserably when Z is already at 0 and I try to home. For instance, if I Z-home twice in a row, the second time would always fail. 

Here is my klippy log: [klippy.log](https://github.com/KevinOConnor/klipper/files/3383626/klippy.log)

I have been using klipper for a while now and it worked well, I'm a happy user. Today when I upgraded the firmware for the first time after several months the issue appeared (I git pull'ed, make menuconfig, make, make flash and restarted the klipper service).

KevinOConnor, 2019-07-11:
The attached log doesn't show any kind of failure.

What do you mean by "fails" - does it continue moving, does it report an error, does it not proceed at all?

We'll need to see a log from the event - if possible, try to reproduce the issue, send an M112 as soon as it occurs, and then attach the log.

-Kevin

urish, 2019-07-11:
Thanks for responding so quickly! 

By fails I mean - it tries to move down for a short while (about 1-2 seconds), while it physically can't (so you hear the stepper clicking). No error is reported. After that, the Z offset is no longer correct - if I try to print, the printer will smash the head into the bed.

Here is the log after reproduction:
 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3383839/klippy.log)

Also, the output from console (printer is homed, then I ask it to home Z again):
```
Send: QUERY_ENDSTOPS
Recv: x:TRIGGERED y:TRIGGERED z:TRIGGERED
Recv: ok
Send: M105
Recv: ok B:32.1 /0.0 T0:38.1 /0.0
Send: G91
Recv: ok
Send: G28 Z0
Recv: ok
Send: G90
Recv: ok
Send: M105
Recv: ok B:31.7 /0.0 T0:37.9 /0.0
Send: M105
Recv: ok B:31.5 /0.0 T0:37.4 /0.0
Send: M112
Recv: // Klipper state: Shutdown
Send: N3 M112*34
Send: N4 M104 T0 S0*37
Send: N5 M140 S0*96
Recv: ok
Closing serial port due to emergency stop M112.
Connection closed, closing down monitor
Changing monitoring state from "Operational" to "Offline (Error: Closing serial port due to emergency stop M112.)"
```

KevinOConnor, 2019-07-11:
I think this is a defect introduced by PR #1766.  Can you retry with new code:

```
cd ~/klipper ; git fetch ; git checkout origin/work-homing-20190711 ; sudo service klipper stop ; make ; make flash ; sudo service klipper start
```

It is necessary to reflash the micro-controller code.

If it still fails, could you reattach the log again.  If possible, try to issue the M112 within 1-2 seconds of the event.

-Kevin

urish, 2019-07-12:
Amazing Kevin, this did the trick!

I tried to homing my Z-axis multiple times in a row, and now it works perfectly, even when my Z end stop is triggered. 😍

KevinOConnor, 2019-07-12:
Thanks.  I've pushed this change to the master branch (commit 7fefe52d).

-Kevin

urish, 2019-07-12:
Awesome! Thank you so much Kevin

closed 2019-07-12 by urish