Klipper3d/klipper #1806: Ender 3 MCU 'mcu' shutdown: Can't assign oid

se7enblade, 2019-07-12:
Installed Klipper on an Ender 3 with BLTouch but no matter what I do the configs fail, even from known working .cfg's and status brings up:

MCU 'mcu' shutdown: Can't assign oid

Worst part is I have had Klipper running before but abandoned as it was still early days with BLTouch/clones.
I've just spent four hours last night and another three this morning googling and trawling through reddit and github without success.
log file enclosed and compressed

se7enblade, 2019-07-12:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3384720/klippy.log)
To keep bot happy, apparently it's blind

KevinOConnor, 2019-07-12:
Your micro-controller keeps resetting when the host tries to configure it.  Try updating to the latest very code from today (see the FAQ for upgrade details) and reflash the micro-controller.  Also, double check that you have an atmega1284p (and not some other type of avr chip).

-Kevin

KevinOConnor, 2019-07-29:
I'm going to close this issue as it appears the conversation has concluded.

-Kevin

closed 2019-07-29 by KevinOConnor

Jufik, 2020-05-10:
I'm having the exact same issue.
I've been using Klipper with my EnderPro3, work(ed) like a charm.

I've received a [BLTouch kit](https://www.amazon.fr/gp/product/B07T4GFCVC/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1), and installed it.

Eventually, the kit comes with a USBISP drive, and I've flashed a bootloader. ([See instructions](https://www.reddit.com/r/ender3/comments/cfmbdy/howto_installing_a_bootloader_to_an_ender_3_pro/))
I've tried Official firmware provided by Creality, and it works fine.

Now, I just want to keep using Klipper, as it gives clear advantages.
*But*, since I've installed the BLTouch, Klipper behaves badly.

In the RPI side, I've redone the install procedure from scratch,
Without changing the stock Ender Config. 
Flashing happens smoothly, but boy, the result is odd.

The BLTouch kit comes with an  Pin 27 Board  to plug BLTouch in, and my guess is that it's what is messing up the config': The screen now is messy.

Removing the [display] from my config' seems make things a bit better, and my guess is there's some mix up on the pins', but boy... that's beyond my knowledge.