Klipper3d/klipper #1808: 2000mm Z shutdowns MCU, again(

aegelsky, 2019-07-12:
Hi Kevin,

it worked before with homing_step_delay tweak, though it took big delay (really big, up to whole minute while Z homing), but after this commit 
https://github.com/KevinOConnor/klipper/commit/dd34768e3afb6b5aa46885109182973d88df10b7#diff-98ab15f13c873597d781fbe4202345ba
that config is gone and now I can use only 200mm out of 2000mm of my Z :(
right now with 2000mm set in config it hangs on Z homing, in log it shows "MCU shutdown", though it's strange, Z axis MCU is called zboard in config.
With changing it to 200mm it works fine.
Also changed Z endstop to probe and inverted it's config, but it looks like not the case, with 200mm it works.

How can I workaround this issue?
Or can you point any supported by klipper motherboard (cheaper is better), that could handle 2000mm Z without tweaks?
Now it's mega/ramps and I'm quite happy with it with klipper and ramps, I was able to start printing on my "monster".

Attaching log and a picture of printer, just for fun :)
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3386793/klippy.zip)
Now I'm stuck at around 1m height (was doing some changes there with Z probe) and have to go down by 20cm(
[image]

Best regards and let me know if you need more info,
Alexander

aegelsky, 2019-07-12:
small update, found couple different errors in log:
MCU 'mcu' shutdown: Command request
MCU 'zboard' shutdown: Timer too close
not sure which one happens first, it looked to me like first one,
while second looks like error from before when tried to set 2000m without tweaking homing_delay

KevinOConnor, 2019-07-12:
Impressive!

After PR #1766 tweaking HOMING_STEP_DELAY should no longer be necessary.  However, it looks like your very large Z axis is causing some host cpu delays.  Can you retry with newer code:
```
cd ~/klipper ; git fetch ; git checkout origin/work-homing-20190712 ; sudo service klipper restart
```

Let me know if that helps (success or failure).
-Kevin

aegelsky, 2019-07-12:
Thank you for quick help, it looks working!
Got error after first successful homing, something like "too many consecutive timeouts", though it homed OK, but then disconnected.
Attached log, but can't find that error there.
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3387361/klippy.zip)
Second homing went OK.

Got couple questions:
 should I stay on that origin at git?
 should I ever re-flash MCU's after these changes? You didn't write about it, but I re-flashed it anyway.

KevinOConnor, 2019-07-12:
Okay, thanks.  I committed that change to master (commit 977d0cf7).

> Got error after first successful homing, something like "too many consecutive timeouts", 

That sounds like an OctoPrint error.

> should I stay on that origin at git?

The change is now on the master branch, so best to pull the latest code and use that.

> should I ever re-flash MCU's after these changes?

There is no harm in re-flashing the mcus.  This particular change was in the host software, so flashing the mcu was not necessary.  Klipper will usually tell you if you need to reflash the mcu.  There's some details in the FAQ: https://www.klipper3d.org/FAQ.html#how-do-i-upgrade-to-the-latest-software

-Kevin

KevinOConnor, 2019-07-13:
Looking at this a little further, I think there is a better way to handle this.  Could you verify everything still works okay with the latest code at:

```
cd ~/klipper ; git fetch ; git checkout origin/work-homing-20190712 ; sudo service klipper restart
```

-Kevin

aegelsky, 2019-07-13:
Made this, looks working. Got a bunch of changes to make, will write update when I'm done, today or tomorrow.

aegelsky, 2019-07-16:
Did a lot of testing last days, no errors so far.
I'm still at origin/work-homing-20190712

KevinOConnor, 2019-07-18:
Okay - I've pushed that change to the master branch (commit bf233029).

-Kevin

KevinOConnor, 2019-08-05:
I'm going to close this issue as it appears the original issue has been fixed.

-Kevin

closed 2019-08-05 by KevinOConnor