Klipper3d/klipper #1809: Non-random layer banding on multiple printers

drphil3d, 2019-07-12:
I've noticed this recently on two of my printers with identical hardware. Printing from Cura 4.1
[klippy.log](https://github.com/KevinOConnor/klipper/files/3387710/klippy.log)

[image]

elliotf, 2019-07-13:
@drphil3d That looks like layer banding, which is usually a physical problem.

what do you have afor a Z axis? Is it a leadscrew? If so, what is the pitch of the leadscrew?

If it is a belted Z, perhaps your pulleys are out-of-round?

See also https://www.evernote.com/shard/s211/client/snv?noteGuid=701c36c4-ddd5-4669-a482-953d8924c71d&noteKey=1ef992988295487c98c268dcdd2d687e&sn=https%3A%2F%2Fwww.evernote.com%2Fshard%2Fs211%2Fsh%2F701c36c4-ddd5-4669-a482-953d8924c71d%2F1ef992988295487c98c268dcdd2d687e&title=Taxonomy%2Bof%2BZ%2Baxis%2Bartifacts%2Bin%2Bextrusion-based%2B3d%2Bprinting

drphil3d, 2019-07-14:
Oh that's really interesting. I have a 4-start M8 threaded rod with the Z nut made from delrin.

drphil3d, 2019-07-14:
I was having some strange issue with git that wasn't allowing me to update to the latest version so removed the directory and reinstalled it, then enabled PID control of the heated bed which seems to have greatly improved things. Thanks

closed 2019-07-14 by drphil3d