Klipper3d/klipper #1810: [NEO PIXEL] Unknown command: neopixel_send

Zwaubel, 2019-07-13:
Just came across the new neo pixel module and immediately had to test it. Unfortunately it did not work. It looks like the module is incomplete/erroneous, since I got the following error although I flashed my µC with the newest firmware (make clean; make; make flash):

Unknown command: neopixel_send
This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer
Klipper state: Not ready

[klippy.log](https://github.com/KevinOConnor/klipper/files/3388803/klippy.log)

KevinOConnor, 2019-07-13:
That code isn't supported on the AVR chips right now - due to the slow speed of the AVR chips, a custom micro-controller implementation would be needed on that platform.

-Kevin

Zwaubel, 2019-07-13:
Okay, thanks for clarifying.

closed 2019-07-13 by Zwaubel

KevinOConnor, 2019-08-07:
Can you retry with the latest code (commit f6d1ab2e)?

-Kevin

Zwaubel, 2019-08-07:
Neopixel control works perfectly on my MEGA2560. Thank you so much!

FYI:
I'm using a single WS2812b pixel to represent a the status of the printer and as of right now, I have not tested anything other than controlling the pixel.
Will do some more in depth testing tomorrow, if required.