Klipper3d/klipper #1812: SAVE_CONFIG small bug [minor]

aegelsky, 2019-07-14:
Hi Kevin

It's not important, just question or FYI.
I've sent "SAVE_CONFIG" to octo terminal, z_offset in printer.cfg got commented and some commented too lines were added to the end of file.
Also I had made "printer (copy).cfg" file in the same dir, not sure it matters, but saved cfg just in case before running save_config.

Here is diff:
pi@raspberrypi:~ $ diff printer.cfg "printer (copy).cfg" 
70c70
< #z_offset: 0
---
> z_offset: 0
141,145d140
< #*# <---------------------- SAVE_CONFIG ---------------------->
< #*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
< #*#
< #*# [probe]
< #*# z_offset = 3.680

And attaching two files, printer.cfg is modified by save_config here.
Not sure if it's a bug, not a big deal anyway.
[printer.zip](https://github.com/KevinOConnor/klipper/files/3389851/printer.zip)

Thank you for Klipper, I like it more and more, fits my needs almost perfectly.

aegelsky, 2019-07-14:
same thing was after saving config after hotend pid calibration

pi@raspberrypi:~ $ diff printer.cfg "printer (copy).cfg" 
40,43c40,43
< #control: pid
< #pid_Kp: 22.2
< #pid_Ki: 1.08
< #pid_Kd: 114
---
> control: pid
> pid_Kp: 22.2
> pid_Ki: 1.08
> pid_Kd: 114
70c70
< z_offset = 3.680
---
> z_offset: 0
141,148d140
< #*# <---------------------- SAVE_CONFIG ---------------------->
< #*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
< #*#
< #*# [extruder]
< #*# control = pid
< #*# pid_kp = 39.875
< #*# pid_ki = 1.846
< #*# pid_kd = 215.362

mental405, 2019-07-14:
This is normal and working as intended.

KevinOConnor, 2019-07-21:
I'm going to close this issue as it looks like the original question was answered - it is the normal behaviour.

-Kevin

closed 2019-07-21 by KevinOConnor