Klipper3d/klipper #1813: BED_MESH_CALIBRATE and SET_GCODE_OFFSET

leonmelson, 2019-07-14:
Hi Guys I have setup Klipper on a custom desktop cnc and it works really good.

I am busy creating a PHP cnc interface to make it a little more cnc like then what octoprint provides and I will make this available when it is in a good place.

I was planning to use the SET_GCODE_OFFSET for the part offset like cnc's use G54 and I might still but I have come into a problem and that is why I have come here to ask if there is a way around this.

I am planning on make PCB's and the PCB needs to be leveled really well and on other cnc you can use apps like autoleveller to level the pcb but Klipper has a 3d printing tool that I was planning to use Mesh Bed leveling.

Now comes the problem when I set a SET_GCODE_OFFSET and want to do a BED_MESH_CALIBRATE the Mesh bed leveling still knows were the true HOME ZERO point is.

Is there a way for the Mesh bed leveling to use SET_GCODE_OFFSET as the ZERO POINT?

By the way Klipper is amazing.
And thank you for any help in advance.

Arksine, 2019-07-16:
No.  The calibration procedure does not use gcode and thus does not take gcode offsets into consideration.  By default the mesh generated is relative to position_endstop, however it is possible to offset the mesh to be relative to one of the probed points by setting the 'relative_reference_index' option.  The generated points and their associated indices are logged on startup (in klippy.log).

leonmelson, 2019-07-16:
OK thank you

I was planning on creating an option in the php program to set the min_point: and max_point: x,y in the config to create the part offset based on the location of the part then do the probe.

And then have the program also set the Gcode_offset for the part offset so I can create the Gcode starting at 0,0.

If the mesh bed leveling was done from 50,50 and I set the Gcode offset to X=50 and Y=50 will that mesh apply to the starting job?

Or do you think this might be the wrong approach.

Thank Your for you help.
Leon Melson

Arksine, 2019-07-16:
I see, you don't want to offset Z, you want to offset X and Y.  If I'm understanding this correctly, you only want to probe the portion of the bed where your part is located.

Setting min_point to 50,50 would initiate the calibration process with the tool at 50,50 (relative to zero) if that is what you desire.  What I'm not sure of is why you need to adjust the calibration area, you can probe the entire surface, set your x,y gcode offset to 50,50, and it will still make the correct adjustment (ie: if you issue a G1 X20, Y20,  mesh leveling will lookup Z at 70,70).

leonmelson, 2019-07-16:
Yes that is correct If I want to cnc a peace of wood stock of say 100x100 and it is offset on the spoilboard by 50x50.

Mesh Bed Level min_point = 50,50 and max_point = 150,150 then I set Gcode offset to  XY 50,50 and the mesh leveling will still use the level area of the stock.

So that sound like it will work then so I will start coding on that then.

Thank you for your help
Leon Melson

KevinOConnor, 2019-08-05:
I'm going to close this issue as it looks like the original question was answered (changing the mesh region at runtime isn't currently supported, but it's possible for someone to extend Klipper to do that).

-Kevin

closed 2019-08-05 by KevinOConnor