Klipper3d/klipper #1814: [FR] LCD via raspberry pi gpio

drphil3d, 2019-07-14:
I know you have been putting a lot of work into supporting lcd controllers via the printers gpio, but I'm wondering if it would be possible to also support using the raspberry pi gpio. There is a plugin for octoprint that displays the print progress, but thats all it does. https://plugins.octoprint.org/plugins/LCD1602_I2cdisplay/

KevinOConnor, 2019-07-21:
If someone wishes to implement this, then that's great.  FWIW, if the display uses spi or i2c then I think it may be possible to do this today by using the existing "linux mcu" code on the Raspberry Pi.  If one is looking to implement one of the "bit banging" displays (such as the hd44780 or st7920) then I'm not sure the RPi would be a good choice (due to timing requirements in those protocols).

-Kevin

drphil3d, 2019-07-28:
oh very interesting, what all is accessible via "linux mcu"

KevinOConnor, 2019-08-05:
There isn't much in the way of documentation for the Linux support.  The best available right now is to look through the src/linux/ directory.

-Kevin