Klipper3d/klipper #1815: Klipper/DWC install onto RPi/Arch linux

Scottbg1, 2019-07-14:
I've tried for the last few days to install klipper and the DWC UI onto a RPi 3b+ running Arch linux.
I have it installed successfully on a Raspbian build but I am a bigger fan of Arch.

I have the packages installed and the klipper service is loaded and running.
I successfully flashed klipper to the MCU, but the printer shows no sign of it being connected to the server. Whenever I flash the MCU when I am using Raspbian the printer's display will show temps and the usual info. When trying to flash when building on Arch, the display only shows white boxes as if no firmware is loaded at all.
I cannot connect to the DWC UI despite me feeling pretty confident that I have the paths set correctly in the conf. 

I don't expect much support in this hub for DWC problems, but I was hoping someone could lend a suggestion that I have Klipper installed properly and that may lead to DWC working as well.
I have searched the 4 corners of the internet with no luck.
Stephan3's github has vague instructions for installing on various linux distros and the only thing he touched on concerning Arch was when he mentioned this:

> Maybe you´ll need to change the startup system for klipper to access ~/klipper/klippy/klippy.py

I would imagine this might be my problem, but what would I need to edit for Klipper to access this path?

Thanks in advance.

Scottbg1, 2019-07-15:
I found one problem.
I had the wrong version of python-jinja installed. 
Not that it made any difference other than fixing that particular log error.
Here is the latest log:

`Config error
Traceback (most recent call last):
  File "/opt/klipper/klippy/klippy.py", line 134, in _connect
    self._read_config()
  File "/opt/klipper/klippy/klippy.py", line 129, in _read_config
    m.add_printer_objects(config)
  File "/opt/klipper/klippy/toolhead.py", line 548, in add_printer_objects
    kinematics.extruder.add_printer_objects(config)
  File "/opt/klipper/klippy/kinematics/extruder.py", line 250, in add_printer_objects
    pe = PrinterExtruder(config.getsection('extruder'), 0)
  File "/opt/klipper/klippy/kinematics/extruder.py", line 19, in __init__
    self.heater = pheater.setup_heater(config, gcode_id)
  File "/opt/klipper/klippy/heater.py", line 253, in setup_heater
    self.heaters[heater_name] = heater = Heater(config, sensor)
  File "/opt/klipper/klippy/heater.py", line 52, in __init__
    algo = config.getchoice('control', algos)
  File "/opt/klipper/klippy/configfile.py", line 64, in getchoice
    c = self.get(option, default)
  File "/opt/klipper/klippy/configfile.py", line 53, in get
    return self._get_wrapper(self.fileconfig.get, option, default)
  File "/opt/klipper/klippy/configfile.py", line 31, in _get_wrapper
    v = parser(self.section, option)
  File "/usr/lib/python2.7/ConfigParser.py", line 340, in get
    raise NoOptionError(option, section)
NoOptionError: No option 'control' in section: 'extruder'
`

drphil3d, 2019-07-15:
You should probably make sure the paths all match what it’s expecting for
raspbian such as the user name. These are just random ideas. Have you
already used Klipper firmware on this printer or is this your first
exposure to it? If your looking for a different operating system I would
highly recommend using Dietpi
-- 

Philip Mally
3D Printing & Rapid Prototyping Services
530-205-3210
pmally@gmail.com

Scottbg1, 2019-07-15:
Yes, that's actually the first thing I checked were the paths. Not saying I missed some but I corrected the obvious ones.

This is my 3rd use of klipper on this printer. The first being along side octoprint. Then I discovered DWC and I have a perfectly fine version of klipper /DWC (No octoprint) running on raspbian now. It's sitting on an SD card and I just have to put it in the pi and reflash the firmware to the printer and I'm in business. I have 4 other RPi's handing various jobs around the house such as a cnc router controller and a media center, etc all running Arch.
I don't have to get it working, I'd just really, really like to.

Stephan3, 2019-07-15:
Please show us the startup log(maybe without the mcu stat noise) after you fixed the jinja thing?

My machines also run arch so it will work for you too

Scottbg1, 2019-07-15:
Hi Stephan,

Forgive my ignorance, but can I assume the startup log are the actions documented _before_ the MCU configs in klippy.log? There isn't much there that I can remember, and I thought the relevant stuff came after the MCU configs. I'll be home in a couple of hours and I will post it asap.

Thank you for your attention to this!

Scottbg1, 2019-07-15:
I wasn't sure exactly which part you needed so I just included the entire klippy.log for this session which includes the MCU configs.

Starting Klippy...
Args: ['/opt/klipper/klippy/klippy.py', '/etc/klipper/klipper.cfg', '-l', '/var/                                                                                                                                                             log/klipper/klippy.log', '-I', '/run/klipper/sock']
Git version: 'v0.7.0-691-g0af5b9ff-archlinux'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.16 (default, Mar 15 2019, 13:03:02) \n[GCC 8.2.1 20181127]'
Start printer at Sun Jul 14 20:42:54 2019 (1563154974.9 11.9)
===== Config file =====
[virtual_sdcard]
path = /home/pi/sdcard

[web_dwc2]
printer_name = Reiner Calmund
listen_adress = 0.0.0.0
listen_port = 4750
web_path = dwc2/web

[stepper_x]
step_pin = PD7
dir_pin = PC5
enable_pin = !PD6
step_distance = .01
endstop_pin = ^!PC2
position_endstop = -20
position_max = 205
position_min = -23
homing_speed = 50

[stepper_y]
step_pin = PC6
dir_pin = PC7
enable_pin = !PD6
step_distance = .01
endstop_pin = ^!PC3
position_endstop = -20
position_min = -23
position_max = 205
homing_speed = 50

[stepper_z]
step_pin = PB3
dir_pin = !PB2
enable_pin = !PA5
step_distance = .0025
endstop_pin = ^!PC4
position_endstop = 0
position_max = 240
homing_speed = 10

[extruder]
step_pin = PB1
dir_pin = PB0
enable_pin = !PD6
step_distance = .0105
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = PD5
sensor_type = ATC Semitec 104GT-2
sensor_pin = PA7
min_temp = 0
max_temp = 250
pressure_advance = 0.4

[heater_bed]
heater_pin = PD4
sensor_type = ATC Semitec 104GT-2
sensor_pin = PA6
control = watermark
min_temp = 0
max_temp = 130

[fan]
pin = PB4

[mcu]
serial = /dev/ttyUSB0

[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 1000
max_z_velocity = 20
max_z_accel = 100

[display]
lcd_type = hd44780
rs_pin = PA3
e_pin = PA2
d4_pin = PD2
d5_pin = PD3
d6_pin = PC0
d7_pin = PC1
=======================
Config error
Traceback (most recent call last):
  File "/opt/klipper/klippy/klippy.py", line 134, in _connect
    self._read_config()
  File "/opt/klipper/klippy/klippy.py", line 129, in _read_config
    m.add_printer_objects(config)
  File "/opt/klipper/klippy/toolhead.py", line 548, in add_printer_objects
    kinematics.extruder.add_printer_objects(config)
  File "/opt/klipper/klippy/kinematics/extruder.py", line 250, in add_printer_ob                                                                                                                                                             jects
    pe = PrinterExtruder(config.getsection('extruder'), 0)
  File "/opt/klipper/klippy/kinematics/extruder.py", line 19, in __init__
    self.heater = pheater.setup_heater(config, gcode_id)
  File "/opt/klipper/klippy/heater.py", line 253, in setup_heater
    self.heaters[heater_name] = heater = Heater(config, sensor)
  File "/opt/klipper/klippy/heater.py", line 52, in __init__
    algo = config.getchoice('control', algos)
  File "/opt/klipper/klippy/configfile.py", line 64, in getchoice
    c = self.get(option, default)
  File "/opt/klipper/klippy/configfile.py", line 53, in get
    return self._get_wrapper(self.fileconfig.get, option, default)
  File "/opt/klipper/klippy/configfile.py", line 31, in _get_wrapper
    v = parser(self.section, option)
  File "/usr/lib/python2.7/ConfigParser.py", line 340, in get
    raise NoOptionError(option, section)
NoOptionError: No option 'control' in section: 'extruder'``

BlackStump, 2019-07-15:
NoOptionError: No option 'control' in section: 'extruder
https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L180

Scottbg1, 2019-07-16:
BlackStamp, I noticed that too but my config file has the control params included, despite what the log states.
Here is the extruder settings from the actual print.cfg:

```
[extruder]
step_pin: PB1
dir_pin: PB0
enable_pin: !PD6
step_distance: .0105
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PD5
sensor_type: ATC Semitec 104GT-2
sensor_pin: PA7
#control: pid
#pid_Kp: 2.151492
#pid_Ki: 0.633897
#pid_Kd: 230.042965
min_temp: 0
max_temp: 250
pressure_advance: 0.4

```

Scottbg1, 2019-07-16:
You know what? I just noticed that those settings are disabled with the '#'.
Let me fix that

Scottbg1, 2019-07-16:
Ok. I eliminated that error but now I am stumped on this one.

```
Config error
Traceback (most recent call last):
  File "/opt/klipper/klippy/klippy.py", line 134, in _connect
    self._read_config()
  File "/opt/klipper/klippy/klippy.py", line 131, in _read_config
    pconfig.check_unused_options(config)
  File "/opt/klipper/klippy/configfile.py", line 235, in check_unused_options
    section,))
Error: Section 'web_dwc2' is not a valid config section
```
I coped and pasted that section straight from Stephan's install instructions.
```
[web_dwc2]
# optional - defaulting to Klipper
printer_name: Reiner Calmund
# optional - defaulting to 127.0.0.1
listen_adress: 0.0.0.0
# needed - use above 1024 as nonroot
listen_port: 4750
#	optional defaulting to dwc2/web. Its a folder relative to your virtual sdcard.
web_path: dwc2/web
```

BlackStump, 2019-07-16:
I think you need to go back to Stephan's instructions, he has done a great job giving step by step instructions.
I think you have missed a step or 2, copy and paste one step at a time

Scottbg1, 2019-07-16:
I just eliminated the entire config section of [web_dwc2] in my .cfg file and my printer display just came to life and the klippy log shows communication with it.
So that config section for DWC seems to be the problem.

BlackStump,
I concur 100% about Stephan's work. I think his efforts to get DWC working with Klipper as an option over Octoprint will further Kevin's efforts and make Klipper even more popular than it's already becoming. Both of these guys are top notch for what they have accomplished.
 But despite your doubts, I have indeed followed his instructions to the letter, hence my success with the Raspbian install. And if I might pat myself on the back, being a novice Linux user,  Debian packages and Arch packages are not named the same and I was able to figure out what was needed to be installed to get this far. 

I just checked my Raspbian install, and the controller commands for the extruder are disabled there too and that install is working flawlessly. So I doubt that particular error was a factor in connecting to the printer considering the former. 

And the be fair, if you read my previous post about the missing ' python-jinja' package, that isn't even in the dependencies list. So please don't question whether or not I followed the instructions properly.
In fact, I did indeed copy and paste the config section from Stephan's instructions on entering the DWC config section and thats where the trouble seems to be, taking into consideration my discovery that Klipper now sees my printer since eliminating that section from the .cfg.

Once I get this resolved I would really like to post step by step instructions on installing Klipper/DWC to an Arch environment so maybe I can save someone some trouble. I would think the developers would appreciate things like that to help relieve questions like I am asking now, so they can concentrate on developing and not trouble shooting install problems.

BlackStump, 2019-07-16:
well yes because of my above comment, if you want to use DWC you must follow the instructions on Stephans page
ie
https://github.com/Stephan3/dwc2-for-klipper
and in particular you have missed this step at least 
ln -s ~/dwc2-for-klipper/web_dwc2.py ~/klipper/klippy/extras/web_dwc2.py
but as I said follow his instructions step by step

Scottbg1, 2019-07-16:
```
[scott@alarmpi ~]$ ln -s ~/dwc2-for-klipper/web_dwc2.py ~/klipper/klippy/extras/web_dwc2.py
ln: failed to create symbolic link '/home/scott/klipper/klippy/extras/web_dwc2.py': File exists
```
Clearly this is a step I have not missed.

BlackStump, please don't do that. 
Don't assume what I have or have not done. I may not have the same knowledge of Linux as you, but I am trying to learn. Your comments have not helped me in the least. I would rather you just not comment at all and leave my post for help completely unanswered than to have someone suggest that I go back and repeat the same instructions that I already have myself a half a dozen times before I relented and reached out for help. I thought that was what this site was for. Apparently I am mistaken. If you have an idea of what my problem is then I would appreciate some insight into how to solve it with direct instructions to try. As you did with creating the soft link to the python file. But without the ambiguously condescending  assumption that I haphazardly skipped the third line of three lines of commands in a code window when I was "copying and pasting" the commands during the install. 

I know that DWC isn't supported here. I stated that in my first post, but I didn't know if the problem wasn't with Klipper itself. And I still don't. And I am betting that you don't either. I will gladly post this on Stephan's hub if you promise not to respond.

BlackStump, 2019-07-16:
ok  I am trying to help you.
You have 2 issues one is klipper and you are in the right site for klipper help and you have DWC issues which is not the right place here.
I made mention of missed  steps  because of the error message in your log
"Error: Section 'web_dwc2' is not a valid config section"
the reason usually is because web_dwc2.py is not in klippy/extras.
but you are absolutely right I am not helpful to you so I am out.

Scottbg1, 2019-07-19:
I finally found some help on another site and a guy named Wilriker assisted me with getting it going.
It seemed to be a permissions problem with the sym-link between the 2 web_dwc2.py files that we couldn't get sorted out.

He suggested putting the dwc2-for-klipper directory in the /opt directory along with the klipper directory and it works perfectly now. He even went as far as writing an install script for the DWC bits that I plan to use to create a PKGBUILD similar to the install script for Raspbian to help other people out that might be having issues getting it to work.

So this indeed was NOT a klipper issue. My apologies for posting here.

closed 2019-07-19 by Scottbg1