Klipper3d/klipper #1816: Install fails

lar3ry, 2019-07-15:
Console output for attempted install. This was on a fresh Octopi install.

pi@octopi:~ $ git clone https://github.com/KevinOConnor/klipper
Cloning into 'klipper'...
remote: Enumerating objects: 15115, done.
remote: Total 15115 (delta 0), reused 0 (delta 0), pack-reused 15115
Receiving objects: 100% (15115/15115), 10.00 MiB | 3.69 MiB/s, done.
Resolving deltas: 100% (11206/11206), done.
Checking out files: 100% (1521/1521), done.
pi@octopi:~ $ ./klipper/scripts/install-octopi.sh

###### Running apt-get update...

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

[sudo] password for pi:
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Get:4 http://archive.raspberrypi.org/debian stretch/main armhf Packages [221 kB]
Get:5 http://archive.raspberrypi.org/debian stretch/ui armhf Packages [45.0 kB]
Err:3 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
  Hash Sum mismatch
  Hashes of expected file:
   - Filesize:11662080 [weak]
   - SHA256:a036d099327c871ba722cea0187061525491c81fe5aaeeb48a545e0de8b36b25
   - SHA1:1ab18738896a891f6b6de1859e629abf0294deba [weak]
   - MD5Sum:fc13d3b6c5064a2bfea645e80e623f2d [weak]
  Hashes of received file:
   - SHA256:a53f4d8d6d964fb260439791dadd2fb20ac449810f036b912a777eeabf403b33
   - SHA1:b8d1ce4686e78a94efc31c8047277daad6516aa9 [weak]
   - MD5Sum:2624ae59e61b08447a5fc0c907bc130e [weak]
   - Filesize:7807353 [weak]
  Last modification reported: Sat, 13 Jul 2019 17:56:35 +0000
  Release file created at: Mon, 15 Jul 2019 06:42:50 +0000
Get:6 http://raspbian.raspberrypi.org/raspbian stretch/non-free armhf Packages [95.5 kB]
Fetched 8,209 kB in 1min 7s (122 kB/s)
Reading package lists... Done
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/stretch/main/binary-armhf/Packages.xz  Hash Sum mismatch
   Hashes of expected file:
    - Filesize:11662080 [weak]
    - SHA256:a036d099327c871ba722cea0187061525491c81fe5aaeeb48a545e0de8b36b25
    - SHA1:1ab18738896a891f6b6de1859e629abf0294deba [weak]
    - MD5Sum:fc13d3b6c5064a2bfea645e80e623f2d [weak]
   Hashes of received file:
    - SHA256:a53f4d8d6d964fb260439791dadd2fb20ac449810f036b912a777eeabf403b33
    - SHA1:b8d1ce4686e78a94efc31c8047277daad6516aa9 [weak]
    - MD5Sum:2624ae59e61b08447a5fc0c907bc130e [weak]
    - Filesize:7807353 [weak]
   Last modification reported: Sat, 13 Jul 2019 17:56:35 +0000
   Release file created at: Mon, 15 Jul 2019 06:42:50 +0000
E: Some index files failed to download. They have been ignored, or old ones used instead.

lar3ry, 2019-07-16:
The bot is complaining about there being no log file. But of course, there is no log file because Klipper did not install. Not sure if you don't see this issue if it has no log file, so I am pasting a dummy log below. I got it from another issue, so you can ignore it.

Unhandled exception during connect Traceback (most recent call last): File "/opt/klipper/klippy/klippy.py", line 134, in _connect self._read_config() File "/opt/klipper/klippy/klippy.py", line 129, in _read_config m.add_printer_objects(config) File "/opt/klipper/klippy/toolhead.py", line 547, in add_printer_objects config.get_printer().add_object('toolhead', ToolHead(config)) File "/opt/klipper/klippy/toolhead.py", line 274, in __init__ self.printer.try_load_module(config, "idle_timeout") File "/opt/klipper/klippy/klippy.py", line 116, in try_load_module self.objects[section] = init_func(config.getsection(section)) File "/opt/klipper/klippy/extras/idle_timeout.py", line 113, in load_config return IdleTimeout(config) File "/opt/klipper/klippy/extras/idle_timeout.py", line 24, in __init__ gcode_macro = self.printer.try_load_module(config, 'gcode_macro') File "/opt/klipper/klippy/klippy.py", line 110, in try_load_module mod = importlib.import_module('extras.' + module_name) File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module __import__(name) File "/opt/klipper/klippy/extras/gcode_macro.py", line 7, in <module> import jinja2 ImportError: No module named jinja2

FHeilmann, 2019-07-16:
your apt can't grab the package indexes (basically your `apt-get update` fails), which suggests that your pi cannot access the internet properly.

lar3ry, 2019-07-16:
Thanks for the reply.
Perhaps you could answer a question for me.
Can I just issue the ./klipper/scripts/install-octopi.sh again without reflashing the octopi image?

lar3ry, 2019-07-16:
oops... Never mind. I just realized that the klipper install parts never did run, so I just did an apt-get update again, and it worked. Running the install part now.

Sorry to bother you.

closed 2019-07-16 by lar3ry