Klipper3d/klipper #1817: Question: two different  mcu ?

cosminr86, 2019-07-16:
Hi! I am currently using a 8 bit mks gen l board and I have a spare 32 bit skr board (I am not using because it has few extra i/o pins). 
I am planning to upgrade the printer to dual z and dual extruder. 
It would be ok to use skr board for x, y, e1, e2, and mks board for z1, z2 and for different i/o ?

technoplunk, 2019-07-16:
This is possible. Note that I think endstops have to be hooked to the same board as the axis motor and I believe temp probes have to be on the same board as their matching heater.

Hywelmartin, 2019-07-17:
It's only the endstops that need to be on the same board as the corresponding stepper drivers... Not heater/temp

cosminr86, 2019-07-17:
I am asking if it's a problem using 2 different architectures 8 bit vs 32 bit

Hywelmartin, 2019-07-17:
No problems...

closed 2019-07-23 by cosminr86