Klipper3d/klipper #1818: "samples" invalid after update

shaiss, 2019-07-17:
I just updated to the latest master with the directions in the FAQ and klipper complained about the 3 following "sample" statements.  However, these still exist in the extras.cfg example.  Were these options removed in a recent update?

Option 'samples_result' is not valid in section 'screws_tilt_adjust'
Option 'samples' is not valid in section 'screws_tilt_adjust'
Option 'samples' is not valid in section 'z_tilt'

[klippy.log](https://github.com/KevinOConnor/klipper/files/3401962/klippy.log)

shaiss, 2019-07-17:
I'll leave this here incase others run into it, but the fix is @ https://github.com/KevinOConnor/klipper/blob/be2d0abbab03667404c784f8543df858af82e7fe/docs/Config_Changes.md

> 20190606: The "samples", "samples_result", and "sample_retract_dist" config options have been moved to the "probe" config section. These options are no longer supported in the "delta_calibrate", "bed_tilt", "bed_mesh", "screws_tilt_adjust", "z_tilt", or "quad_gantry_level" config sections.

upon rechecking the example.cfg, the sample value has indeed been removed from z_tilt and screws_tilt_adjust

so this was a user error :O

closed 2019-07-17 by shaiss

VanKurt, 2019-11-13:
I moved it to the probe section like this:
```
[probe]
samples: 2
```

This gets me the following error:

> Option 'samples' is not valid in section 'probe'

Nitrosito, 2019-11-16:
Me too Firmware version: v0.8.0-114-gd38aed07

marco123445, 2019-12-02:
@KevinOConnor  I am a bit confused now where is the samples section now? Iam not able to find it the example.cfg is not updated yet i guess.