Klipper3d/klipper #1819: Temperature stability issues

urish, 2019-07-17:
I'm printing with PETG at 235°C, and I'm observing the following: The temperature is pretty stable for few first few minutes of the print, then it starts fluctuating. Here is a screen shot from Octoprint showing the issue:

[image]

Eventually, for longer prints, they sometimes fail with "Heater extruder not heating at expected rate" (in the case of the graph above, the print failed with this message about 5 minutes after the screen shot was taken).

I've tried the PID calibration for the heater (calibrated for 235°C), but am still seeing this kind of fluctuation. Would appreciate any points on how to further investigate. Here is my klippy log file (the last print there is the one that failed with the heather error):

[klippy.log](https://github.com/KevinOConnor/klipper/files/3402215/klippy.log)

cosminr86, 2019-07-17:
What main board do you have?

urish, 2019-07-17:
CR-20 with Creality V2.1 main board:

[image]

urish, 2019-07-17:
So far, I compared printing with heated bed and without heated bed, and it seems like the hotend temperature is much more stable when heated bed is off:

[image]

Both prints were done without actual filament in the extruder (just to save on plastic)

strangely enough, when I turn both the hotend and heated bed on manually, not during print, it seems to keep the temperature pretty stable:

[image]

stegmanno1, 2019-07-17:
Have the same problems here, after some time of printing the temperature suddenly is not stable...
[image]

Soon after the print cancels with error "Heater extruder not heating at expected rate" and printing stops.
If i manually increase the temperature a little when it beginns to swap, the print continues without a problem...
[image]

urish, 2019-07-18:
Okay, this is very strange - I took [one of my GCode files](https://github.com/KevinOConnor/klipper/files/3405277/CCR10_PiCam_U_Bracket.gcode.txt), 
 and manually sent the commands from the beginning of the file to the printer:

```
M140 S70
M105
M190 S70
M104 S235
M105
M109 S235
```

The printer heats up and keeps the temperature stable for 10 minutes without a problem:

[image]

However, when printing the, temperature starts to fluctuate after 3-4 minutes or so (that's the graph on the right side): 

[image]

Finally, after canceling the gcode print, I ran the first experiment again (sending the heather gcode command from the beginning of the file manually), and it seems to be able to keep the temperature stable (graph on the right side):

[image]

@KevinOConnor I'd love to get some ideas how to further debug this...

urish, 2019-07-18:
I did another experiment, adding 5 minute wait (`M4 S300`) inside the gcode file, just after the code that sets the heater temperature:

```
;FLAVOR:Marlin
;TIME:1918
;Filament used: 1.29036m
;Layer height: 0.2
;MINX:85.85
;MINY:97.1
;MINZ:0.2
;MAXX:134.15
;MAXY:122.9
;MAXZ:39.9
;Generated with Cura_SteamEngine 4.1.0
M140 S70
M105
M190 S70
M104 S235
M105
M109 S235
G4 S300
M105
M82 ;absolute extrusion mode
...
```

This seems to delay the temperature fluctuations by several minutes: 

[image]

Later, I tried pausing mid-print and waiting for ~10 minutes to see if the fluctuations stop, but they continue even after pausing (and keeping the heater on, of course):

[image]

KevinOConnor, 2019-07-18:
This looks like line noise on the thermistor wires and/or power supply voltage fluctuations.

To reduce thermistor wire line noise, avoid running the thermistor wires next to the stepper motor and heater wires.  Some people add a capacitor to the thermistors as well.

If the power supply is struggling to keep up, it may help to reduce the power to the heaters - for example, by adding `max_power: 0.3` to the bed_heater section and `max_power: 0.7` to the extruder.

-Kevin

Lissanro, 2019-07-18:
I had similar issue, it turned out to be hardware problem. Recently I changed nozzle in my extruder and after calibrating PID for extruder for target 220 this is what I was getting when heating the extruder to 190 (30 degrees below target, this made the issue more pronounced):
[image]
When I tried to use watermark control I was getting this:
[image]
In my case the issue turned out to be that ground connection between Maple Mini and RAMPS wasn't good enough - I accidentally discontented one of ground wires and this caused more noise in the thermistors. For me the issue was fixed by improving ground connection (in my case by connecting the wire I accidentally disconnected). But there maybe other reasons why there is too much noise in the thermistors even if ground connection is good especially if thermistor wires are long and unshielded.

urish, 2019-07-18:
Thanks for all the input!

As far as power supply - I connected it to a multimeter, and it seems like the voltage is fluctuating between 3.95v and 4.2v while printing - this happens both while the temperature is stable and when it start fluctuating. The idle voltage of the printer is 24.2, and then voltage drops to 3.96v when I ask the printer to heat both the nozzle and the bed at the same time. Does this sound like the expected behavior?

Lissanro, 2019-07-19:
If you have 24V power supply and under load your multimeter shows you as little as 3.95V or 4.2V, then something is very wrong. Perhaps some capacitors failed in your power supply and do not filter switching noise anymore under load - in this case typical multimeter will show low voltage because of high frequency noise, even if real voltage is still sufficiently high to drive heaters. Obviously, so much noise in power supply is not good and is very likely to cause issues.

When voltage is actually low, this is different issue. For example, recently I had a problem with power supply of my printer - normally it was 12V but under load it became 8-9V and my printer was still working but bed could not reach high temperatures to print ABS. There was no high frequency noise in this case - the voltage was really this low. After I replaced power supply everything started to work normally - under load it was almost 12V. I think that 24V power supply under load should maintain voltage within 22V-24V range and high frequency noise should be filtered good enough so you can see 22V-24V under load on your multimeter.

Another possible cause of noise in thermistors is bad ground connection (or lack of it) in power supply. You can check this by switching your multimeter to measure AC voltage and touching power supply's metal case with one probe of your multimeter while the other probe touches nothing or your body - if you notice AC voltage like few volts or more, ground connection of power supply is bad. This actually can cause issues including additional noise in thermistors.

urish, 2019-07-19:
@Lissarno thanks for your very detailed comment! I just noticed I wrote `3.95` and `4.2` instead of `23.95` and `24.2`, I was probably too tired and frustrated at the time when I wrote this comment. I have edited it now to reflect the actual numbers. Sorry for the confusion!

Lissanro, 2019-07-19:
My suggestion to check if power supply grounded properly is still relevant. I know from experience that 3D printer with ungrounded power supply may work unpredictably.

You previously mentioned that with bed turned off temperature appears to be more stable and that suggests you have hardware issue. If your power supply is grounded properly then issue is caused by something else but finding exact cause is not always easy so you may want to try some workarounds, for example KevinOConnor already suggested to try to limit max power or add capacitor to the thermistor.

urish, 2019-07-20:
Thanks for the suggestion, I will try to check the ground of the PSU!

urish, 2019-07-21:
Update: I have been printing in lower temperatures since (200-220deg), and it seems like the temperature consistency is much better at this range:

[image]

I will close this issue, as it seems to be hardware-related rather than something specific to klipper. Thanks for all the help so far!

closed 2019-07-21 by urish