Klipper3d/klipper #1821: Z axis movement very very slow

alan242, 2019-07-18:
I have a Monoprice Maker Select Plus (Wanhao Duplicator i3 plus) printer. I am using the supplied printer-wahhao-duplicator-i3-plus-2017.cfg file.

I have a problem with the z axis moving very very slowly. If I home the print head, that action appears to work just fine. If use the manual movement to move the head, for example up 10mm, it takes about 2 minutes exactly to complete the operation. Also when printing, the head never seems to move up off the bed to allow filament to come out.  The X and Y axis movements appear to work normally.

I did not have this problem with the stock firmware or the ADVi3++ 4.0.2 firmware. I tried the klipper git version v0.7.0-691-g0af5b9f-dirty from a few days ago and the 0.7.0 release version. Both appear to have this issue.

Any ideas on what might be wrong or need adjusting?

[octoprint terminal.txt](https://github.com/KevinOConnor/klipper/files/3404696/octoprint.terminal.txt)

[klippy.log](https://github.com/KevinOConnor/klipper/files/3404695/klippy.log)
-->

KevinOConnor, 2019-07-21:
Your log indicates you sent a `G1 Z10 F5` which instructs the printer to move at a very slow rate - if you want to move faster, use a move command with a higher speed.

-Kevin

alan242, 2019-07-23:
Ok. That was the clue. I flashed the printer back to ADVi3++ and realized I was using the LCD control to move the print head. With Klipper, this printer does not have LCD support, so I have to use Octoprint’s control panel to move the head. With either firmware and the Octoprint control the head moves slow. I noticed when I use the LCD control to move the head, the command echoed in the terminal has F240. I am not sure where to adjust that in Octoprint. I can however use the terminal to move it fast.

The other problem I found was I needed to change stepper_z position_endstop from 0.5 to 0.1.

A print is running now. Should know the results in a few hours. Thanks!

Alan

KevinOConnor, 2019-08-05:
I'm going to close this issue as it looks like the original question was answered (the slowness was due to the g-code request).

-Kevin

closed 2019-08-05 by KevinOConnor