Klipper3d/klipper #1825: [F R] TMC 2209 DIAG pin

cosminr86, 2019-07-19:
Hi!
It is sensorless homing implementation planed for tmc2209? I notice diag pin is missing from tmc2209 configuration. 

Thank you!

KevinOConnor, 2019-07-21:
There was some work done on it (see PR #1749).  I'm unsure what the current state is.

-Kevin

mental405, 2019-07-21:
The branch needs reworking to get it back in line with master, but not much. There is a register mismatch i  tmc. py in the virtual endstop (en_spreadcycle vs en_pwm_mode) because the name is different on the 2209s. There are a couple of other minor behavioral differences with the drivers and homing. The tmc2209 nust needs the virtual pin created for the endstop to make that part work. Other than that sensorless works great on the 2209s.

Stephan3, 2019-07-21:
Yes. I Know. curently i am on vacation. I will finish that soon™

KevinOConnor, 2019-08-17:
Any further updates on this?  If someone has the hardware and is willing to run tests, I can put together some test code.  Let me know.

-Kevin

cosminr86, 2019-08-17:
I am currently waiting for the code to put the tmc2209 on the mainboard.

If I will get the code I will test it!

Thanks!

Stephan3, 2019-08-17:
2209 does sensorless homing  better in stealthchop, regarding to they’re documentation. I will look into that the next week.

darkernoise, 2019-08-17:
I have 2209s installed on two printers running Klipper already.  I will be happy to test and/or help with code.

KevinOConnor, 2019-08-18:
FYI, I put together some test code:

cd ~/klipper ; git fetch ; git checkout origin/work-tmc2209-20190817 ; sudo service klipper restart

-Kevin

cosminr86, 2019-08-18:
I manage to make sensorless homing working.
The sensorless homing docs have some parts different from tmc2100.
In the docs it says diag1_pin, correct one is diag_pin and the #driver_SGTHRS is correct for the 2209 not driver_SGT how is in the docs. The docs says that driver_SGTHRS should have value between -64 to 63 but I was unable to use a negative value, but a value above 63 was ok. I am currently using 82.

Somehow it had broken my normal homing.
I had changed from 2208 to 2209 in config and the normal homing was not working, maybe something weird with the mainboard  (skr 1.3)

mental405, 2019-08-18:
sgt is different on the 2209s, it is in the datasheets. The 2209s use an unsigned int and the larger the value the more the sensitivity.

darkernoise, 2019-08-18:
Here are some notes and observations from my testing:

Test Environment:
    • Custom-built cartisian printer modeled after the Prusa I3
    • Running on Bigtreetech SKR1.3
    • BIQU TMC2209 V1.1

Naming Convention Questions:
    • I see that the TMC2209 class config parameters have a mixed use of upper and lower case, which is a kind of confusing.  For example:

	set_config_field(config, "PWM_GRAD", 14)
       	set_config_field(config, "pwm_freq", 1)

	Are these case sensitive and/or, does this matter?

    • Again, I don’t know if this matters and I don’t mean to nitpick, but the sample configs have the diag pin in the convention: “diag1_pin” whereas the TMC2209 class the config field is just “diag_pin”.
      
    • Sensorless Homing help will need to be updated since now, the SGTHRS register accepts values from 0 to 255.  Previously, the start value for sensor-less homing was “0”, where now it should be maybe 200?

Testing:
    • Regression tested after this patch to ensure that basic TMC2209 functionality still works : Pass
    • Enabled the config param: diag_pin, and driver_SGTHRS as follows:

Notes from the datasheet:

DIAGis pulsedby StallGuard, when SG_RESULT falls below SGTHRS. It is only enabled in StealthChopmode, and when TCOOLTHRS ≥ TSTEP> TPWMTHRSTthe external motioncontroller should react to a single pulse by stopping the motor if desired. Set TCOOLTHRS to match the lower velocity threshold where StallGuard delivers a good result. 

So here is my first question, I don’t see the register, “TCOOLTHRS”, setable in the config files.  Nor, do I see this register value listed in dump_tmc.

fist test config:

[tmc2209 stepper_x]
uart_pin: P1.17
microsteps: 16
interpolate: True
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250
#sense_resistor: 0.110
diag_pin: PXDIAG1
driver_SGTHRS: 250  # tuning value for sensor-less homing, set to 250 as a start

TMC Dump:

Send: dump_tmc stepper=stepper_xRecv: // ========== Write-only registers ==========Recv: // IHOLD_IRUN: 0008190f IHOLD=15 IRUN=25 IHOLDDELAY=8Recv: // TPWMTHRS:   00000026 TPWMTHRS=38Recv: // TPOWERDOWN: 00000014 TPOWERDOWN=20Recv: // SGTHRS:     000000fa SGTHRS=250Recv: // ========== Queried registers ==========Recv: // GCONF:      000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1Recv: // GSTAT:      00000001 reset=1Recv: // IFCNT:      000000b6 IFCNT=182Recv: // OTP_READ:   0000000d OTP_FCLKTRIM=13Recv: // IOIN:       21000041 ENN=1 PDN_UART=1 VERSION=0x21Recv: // FACTORY_CONF: 0000000d FCLKTRIM=13Recv: // TSTEP:      000fffff TSTEP=1048575Recv: // MSCNT:      000000c8 MSCNT=200Recv: // MSCURACT:   005200e9 CUR_A=233 CUR_B=82Recv: // CHOPCONF:   14030053 toff=3 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1Recv: // DRV_STATUS: c00f0000 CS_ACTUAL=15 stealth=1 stst=1Recv: // PWMCONF:    c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12Recv: // PWM_SCALE:  0001002f PWM_SCALE_SUM=47 PWM_SCALE_AUTO=1Recv: // PWM_AUTO:   001c005d PWM_OFS_AUTO=93 PWM_GRAD_AUTO=28Recv: // SG_RESULT:  0000006e SG_RESULT=110

Test results:
    • tested with SGTHRS at multiple values: result: stall guard is not working.  Thus, I did not setup virtual endstop.

Please confirm, will sensor-less homing work correctly without setting the  TCOOLTHRS register?

Addendum: 

Definitions:
TCOOLTHRS: Lower velocity thresholdfor switching on CoolStepand stall output. Below this velocity CoolStepbecomes disabled (not used in STEP/DIRmode). Adapt to the lower limit of the velocity range where StallGuard2gives a stable result.

Value range: 1...1048575

SGTHRSDetection threshold for stall. The StallGuardvalueSG_RESULTbecomes compared to the double of this threshold. A stall is signaled with SG_RESULT≤SGTHRS*2

Value range: 0...255

KevinOConnor, 2019-08-19:
@cosminr86 - so, did sensorless homing work on the tmc2209 with the work-tmc2209-20190817 branch?

Yes, it does look like the Sensorless_Homing.md document will need an update.

-Kevin

cosminr86, 2019-08-19:
Yes, it works!

KevinOConnor, 2019-08-19:
@darkernoise 

>  I see that the TMC2209 class config parameters have a mixed use of upper and lower case, which is a kind of confusing.

It's done that way because the TMC specs do that.  Yes, it is confusing.  The case is consistent in the example-extras.cfg description of the parameters, however.

> Again, I don’t know if this matters and I don’t mean to nitpick, but the sample configs have the diag pin in the convention: “diag1_pin” whereas the TMC2209 class the config field is just “diag_pin”.

Yes, because the TMC specs define the pin that way.  The tmc2130 has a `diag1_pin` while the tmc2209 has a `diag_pin`.

> So here is my first question, I don’t see the register, “TCOOLTHRS”, setable in the config files.

That parameter is automatically set (and cleared) during sensorless homing (see klippy/extras/tmc.py:TMCVirtualEndstop).

It's not shown in the TMC_DUMP output, because it can not be queried from the chip (it's a write-only register), and the "write-only" section of TMC_DUMP only shows registers that have actually been written.  So, if it was never written, it doesn't show up in TMC_DUMP.

I don't think there's utility in having a user specify a `driver_TCOOLTHRS` - it's something that code would need to be written for.  (At a minimum, to translate a mm/s to the corresponding clock ticks of TCOOLTHRS.)

-Kevin

KevinOConnor, 2019-08-19:
> Yes, it works!

Okay, thanks.  I committed the change to the master branch (commit a482ddf5).  I don't think this change would impact regular endstop detection.

-Kevin

darkernoise, 2019-08-19:
Hi all,

It sounds like the issue here might be just that I am not familiar with the TMC interfaces.  I read the data sheet and I interpreted that SGTHRS is the threshold for sensor-less homing, as well as stallguard.  From there, TCOOLTHRS looks to be the lower threshold for coolstep and stallguard, which is required for the diag_pin.  Is that not correct?

Either way, I did install the patch and set SGTHRS to 200 and 250 as tests.  From there, I triggered a G1 X<value> move and blocked the X carriage.  The carriage did not stop.  This is why I did not proceed to the virtual_endstop setup.  

Again, this could be my ignorance.  Is the above procedure correct or am I missing something?

darkernoise, 2019-08-19:
@cosminr86 Could you share your 2209 settings?  I would like to see your register settings.  Here is mine:

[tmc2209 stepper_x]
uart_pin: P1.17
microsteps: 16
interpolate: True
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250
#sense_resistor: 0.110
diag_pin: PXDIAG1
driver_SGTHRS: 250  # tuning value for sensor-less homing, set to 250 as a start

KevinOConnor, 2019-08-19:
> I read the data sheet and I interpreted that SGTHRS is the threshold for sensor-less homing, as well as stallguard.

Klipper does not support general "stall detection".  It only supports stall detection during homing.  In which case, "stallguard" is effectively the same thing as "sensor-less homing".

> From there, TCOOLTHRS looks to be the lower threshold for coolstep and stallguard

Klipper does not support "coolstep" and thus TCOOLTHRS is not relevant.  Yes, technically, "coolstep" must be enabled during sensor-less homing, but the code should automatically do this.

> Either way, I did install the patch and set SGTHRS to 200 and 250 as tests. From there, I triggered a G1 X move and blocked the X carriage. The carriage did not stop.

Klipper does not support arbitrary "stall detection".  The carriage would only stop if the tmc2209_stepper_x:virtual_endstop is properly configured and only then during a homing operation.

-Kevin

cosminr86, 2019-08-19:
I have driver_SGTHRS to around 80.
But I  do use sensorless home.
Above 85 it gives me an endstop still triggered.

While I did the test i had second home distance to 50 mm. 

I am not sure, but I don't know if stallguard is implemented in klipper yet.

darkernoise, 2019-08-19:
@KevinOConnor: Makes complete sense. Newbie trying to help does not have all the information and gets confused.  Story of my life :)

@Cosminr86: Thanks. I will test with virtual-endstop setup.

mental405, 2019-08-19:
> I have driver_SGTHRS to around 80.
> But I do use sensorless home.
> Above 85 it gives me an endstop still triggered.
> 
> While I did the test i had second home distance to 50 mm.
> 
> I am not sure, but I don't know if stallguard is implemented in klipper yet.

I think you are confusing step loss detection with stallguard.

StallGuard is the feature of the Trinamic chips that measures back EMF in the motor to determine if the motor is about to miss a step. 

The implementation for sensorless homing in Klipper and others involves telling the chip to communicate whether or not it is about to miss a step by pulling the diag pin high (?). The threshold relates to how much "stall" to take before sending that signal.

There exists the ability  to read the StallGuard register directly (at least for the 2209's) so that sensorless homing can be done via the SPI or UART interface instead of relying on the Diag pin but the principle is the same.

I am not sure what has to be done for a step loss module but I opened an issue requesting it some time ago and nobody seemed interested in doing it.

KevinOConnor, 2019-09-06:
I'm going to close this issue as basic tmc2209 sensorless homing should now be supported.

-Kevin

closed 2019-09-06 by KevinOConnor

terrorhai, 2019-11-24:
I will reopen this, usually anyone in this topic could help me with my problem?
https://github.com/KevinOConnor/klipper/issues/2201