Klipper3d/klipper #1826: Tmc2209 printer.cfg

Saso-lab, 2019-07-19:
Trying to install the tmc2209 but it gives the error "tmc2209 stepper_a is not a valid config section". 
Maybe something missing in the printer.cfg related to the section tmc? 
[https://pastebin.com/ZETN2zuK](url)
[klippy.log](https://github.com/KevinOConnor/klipper/files/3411039/klippy.log)

Saso-lab, 2019-07-19:
I updated klipper to the lastest version, and this is the new log 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3411681/klippy.log)
[image]

Stephan3, 2019-07-19:
Drivers need soldering. You soldered the uart bridge. Maybe you want the humans more detail here. What vendor are the drivers? What board are you using etc..

Saso-lab, 2019-07-19:
So the board is a mks gen L, the drivers are TMC2209 by BigTree v1.1
They should be ready for using them in Uart mode without solder anything.
On the top of the drivers there are 2 pin on the small side and 3 on the large side, i connected the middle one of this to a pin on the board...
(i took the image of tmc2208 to better explain the pin)
[image]

jannau, 2019-07-19:
I see the same error with Watterott TMC2209 on a SKR v1.3. Watterott TMC2208 workover UART in the same slot with the same config.

Stephan3, 2019-07-19:
Watterott on skr is Pin missmatch. I made the code with watterott drivers. @jannau
Remove the pins from watterott and make a short wire to the jumper and connect it there. Watterotts needs soldering.

@saso-lab show us the bottom side.

Saso-lab, 2019-07-19:
If it can help you i've done a picture of the bottom in which you can see there is a resistor (not 3 pads to solder as the previous tmc 2208). 
[image]

jannau, 2019-07-19:
@Stephan3 thanks, I looked more than once and did not see that. works now as expected

KevinOConnor, 2019-07-21:
Some things to check with the tmc2209: make sure motor power is on when starting Klipper; make sure the MSx pins are all grounded (to select address zero); make sure the uart pin on the tmc2209 is routed to a pin on the micro-controller; make sure the correct micro-controller pin is in the config file.

-Kevin

Saso-lab, 2019-07-21:
@KevinOConnor thanks for the reply, the motor power is on when Klipper starts (raspberry and PSU share the same electrical socket). The MS pins are all grounded (i'm switching from a4988 1/16 steps ).
I connected the central pin of the driver (image above) to a pin on the mks gen l.
I followed this image to connect the jumpers. For exemple in the printer.cfg (code above) the uart pin for the stepper A (according to the image) is ar40...
[image]

Saso-lab, 2019-07-25:
After several attempts i could use them in uart mode with mks gen l. I bought a skr v1.3 and everything works fine now. I think there was something wrong related to the uart communication with the board (i am sure that pin name and jumper were fine). Anyway i think that this issue (not solved) can be closed...

closed 2019-07-25 by Saso-lab

Lazurx, 2020-01-18:
Hmm. No further investigation?