Klipper3d/klipper #1829: Creality 1.1.5 SilentBoard

timoschuett, 2019-07-22:
Hello, i use Klipper on my Ender3 on stock board without any problems. But it is too loud. Now I ordered the “new” Creality Silent Board which calls it plug and play with the TMC2208 installed.

After that I just read, that this cause problems with Klipper and is “not safe” to use. 
Why is it unsafe? Does it just stop printing with problems or do I put myself in danger with my printer just catching fire??

I was satisfied with Klipper on the stock board and just bought this “plug and play” thing for reducing the noise but still using the fast speed of Klipper. Isn’t this possible? Are there any other plug and play boards which are silent but compactible with Klipper and are plug and play?

Thanks!

KevinOConnor, 2019-07-24:
See the FAQ ( https://www.klipper3d.org/FAQ.html#my-tmc-motor-driver-turns-off-in-the-middle-of-a-print ) and issue #1096 

-Kevin

towe96, 2019-07-28:
The Creality Silent 1.1.5. uses UART - you should be fine, and the warning doesn't apply to the board.

It's just a more expensive version of the SKR Mini E3 boards with worse drivers.

timoschuett, 2019-07-28:
Thank you towe96, so do i have to put the tmc anyhow in the config file?
Then i will flash back Klipper on it later :)

KevinOConnor, 2019-08-05:
This is the first I've heard of the Creality board having tmc uarts routed - do you have the corresponding pin configuration for the uart pins?

-Kevin

timoschuett, 2019-08-05:
No i dont, maybe towe?
Creality told me on fb that they „would not recomment, to let the tmc connect via uart“ so maybe they‘re really wired and just need to connect.

towe96, 2019-08-05:
I strongly believe I read somewhere that the board would support sensorless homing, but I can't find that anymore.
I believe it might not have UART after all. Creality has never really been good at "state of the art" software. It still ships with a botched Marlin 1.1.8. according to their website.

KevinOConnor, 2019-08-17:
I'm going to close this issue as I think the original question has been answered (don't use the tmc2208s in standalone mode with Klipper).

-Kevin

closed 2019-08-17 by KevinOConnor